tzapu/WiFiManager #701: Attempting to load captive portal page results in ESP8266 crash

Merlin04, 2018-08-15:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  I'm not sure, the release is from a few months ago. I can't upgrade it due to modifications to the code (mainly the portal itself). 

Esp8266/Esp32: 

Hardware:			ESP-12f

Core Version:      	2.4.0

### Description

I turn on the device, and connect to the AP it produces. The captive portal window pops up, but nothing loads in it. I try going into a web browser and going to 192.168.4.1, but all I see is the background color (after a long wait). After that, the ESP crashes. Here's the full output:

```
*WM: 
*WM: AutoConnect
*WM: Connecting as wifi client...
*WM: Using last saved values, should be faster
*WM: Connection result: 
*WM: 1
*WM: SET AP STA
*WM: 
*WM: Configuring access point... 
*WM: Device Setup
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root

Exception (29):
epc1=0x4000e1c3 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000018 depc=0x00000000

ctx: sys 
sp: 3ffffc50 end: 3fffffb0 offset: 01a0

>>>stack>>>
3ffffdf0:  00800000 3ff20a00 40231eaf 00000001  
3ffffe00:  402205ee 4022050c 3fff4b14 40100ec6  
3ffffe10:  3ffec965 401042c3 3fff03a0 00000000  
3ffffe20:  00000018 3fff0db0 3fff0d2c 00000000  
3ffffe30:  402206bb 3ff20a00 3fff4b14 3ffedede  
3ffffe40:  3ffedec4 00000000 00000000 00000001  
3ffffe50:  00000017 0e186b1c 401021f2 3ffedec4  
3ffffe60:  3ffedede 0000002c 00000004 401004f4  
3ffffe70:  3ffedece 3fff41d4 4021d210 3ff20a00  
3ffffe80:  3fff4b14 3fff47fc 00000015 4021d0e3  
3ffffe90:  3fffc278 40101f8c 3fffc200 00000022  
3ffffea0:  3ffed204 00000000 000007db 00000000  
3ffffeb0:  40000f68 00000030 0000000e ffffffff  
3ffffec0:  40000f58 00000000 00000020 00000000  
3ffffed0:  00000004 402122d8 00003a98 00000000  
3ffffee0:  3fff1bb8 3ffec774 3fff1bb8 3fffdab0  
3ffffef0:  00000000 3fffdcb0 00000004 00000030  
3fffff00:  40224d49 ffffffa4 00000000 402112b4  
3fffff10:  4021c91f 3ffedec4 3fff03a0 3fff4b14  
3fffff20:  00000000 3ffedece 3ffedf07 3fff37f0  
3fffff30:  4023299d 00000000 00000000 3ffedece  
3fffff40:  40231b48 0000004d 3ffedeb8 3fff41d4  
3fffff50:  3ffedeb8 00000000 00000004 40231472  
3fffff60:  3fff03a0 3ffedeb8 3fffdcc0 3ffecaa0  
3fffff70:  3ffeded4 3fff4dbc 00000040 3ffecaa0  
3fffff80:  00000040 3fff03a0 00000000 3fffdcb0  
3fffff90:  40230d6f 3fffdab0 00000000 40237edf  
3fffffa0:  3ffecaa0 40000f49 3fffdab0 40000f49  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset
rl
```

### Settings in IDE

Module:  			Adafruit HUZZAH ESP8266

Additional libraries: 
Adafruit MQTT
Servo

### Sketch

I am unable to share the sketch because it would take too much time to remove sensitive information from it (it was built around a specific account). The code works on other ESP boards (practically the same as the huzzah, just re-designed to be smaller). The code isn't the problem, but what the code does is it subscribes to an MQTT feed. When it gets a message, the servo rotates. 

I have customized the setup screen to add a logo, that might be taking up a lot of space. 

### Debug Messages
I decoded the stack:
```
0x401004f4: calloc at /home/me/.arduino15/packages/esp8266/hardware/esp8266/2.4.0/cores/esp8266/umm_malloc/umm_malloc.c line 1688
0x402122d8: sys_check_timeouts at core/timeouts.c line 314
0x402112b4: memp_malloc at core/memp.c line 404
```

tablatronix, 2018-08-16:
corrupt flash, too much memory, or flash size set wrong in ide

Merlin04, 2018-08-16:
I checked all the settings and they were correct. Removing the image fixed the issue, but I'm not sure why that was the problem.

tablatronix, 2018-08-16:
did you check your free heap ?