tzapu/WiFiManager #702: wifi doesn't connect after exiting the captive portal with ESP core 2.4.2

andysc, 2018-08-16:
This might be for "upstream", but between ESP8266 core 2.4.1 and 2.4.2, connectWifi comes back with WL_IDLE_STATUS (0) and doesn't connect to the newly set wifi credentials.

With 2.4.1, the portal closes itself correctly and the device connects correctly to the new credentials.

With 2.4.2, the same code leaves the portal window open and gives WL_IDLE_STATUS and doesn't connect.

This is with a Wemos R2 D1 mini device.

tablatronix, 2018-08-16:
what branch?

tablatronix, 2018-08-16:
you can try development branch and post logs

andysc, 2018-08-16:
0.14.0

andysc, 2018-08-16:
(assuming you meant WiFiManager, not ESP core?)

andysc, 2018-08-16:
yes, development branch of WiFiManager works correctly.

Any chance you could backport that part of the code (startConfigPortal) to the current codebase? 

That looks like a huge re-write you're doing in development branch! wow :)

tablatronix, 2018-08-16:
yeah I have no idea what the issue is to fix though

andysc, 2018-08-21:
I'll just go back to the earlier version of the core, as I know that works with the current released WiFimanager.
Keep up the great work :)

andysc, 2018-08-21:
do you "need feedback" from me? #justchecking

tablatronix, 2018-08-21:
I need debug logs of failure

andysc, 2018-08-22:
OK. This is the console output from when I save the new wifi config to where it goes round again (having failed to connect to the new wifi due to the Connection result 0 (WL_IDLE_STATUS).

How do I turn on more useful debug?

```
*WM: VodafoneMobileWiFi-D81949
*WM: -86
*WM: Sent config page
*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 0
*WM: Failed to connect.
*WM: Request redirected to captive portal
*WM: Scan done
*WM: DUP AP: _Free SOU Airport
```

tablatronix, 2018-08-22:
is this a hidden ap ?

If you get a chance turn on esp debugging wifi level

In development you can try
setSaveConnectTimeout(seconds); to increase the connection wait loop, by default it uses waitforconnectresult, maybe there is a bug in there, or problems with your router

tablatronix, 2018-08-22:
connect result is idle_status I think, which is weird

andysc, 2018-08-22:
No, not hidden -  it's the WiFi hotspot on my iPhone - usually works fine.
I tried esp debugging wifi but it just inserted lots of WiFi event 7 into the stream, like this:
```
*WM: MW40V_B9B3
*WM: -75
*WM: Sent config page
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
*WM: WiFi save
*WM: Sent wifi save page
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Connection result: 
*WM: 0
*WM: Failed to connect.
wifi evt: 7
wifi evt: 7
*WM: Request redirected to captive portal
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
scandone
*WM: Scan done
*WM: DUP AP: BWU2
```

Yes, result 0 is WL_IDLE_STATUS, which, yes, is weird.

I'll try changing setConnectTimeout... thanks.

andysc, 2018-08-22:
That didn't help - I set 
`wifiManager.setConnectTimeout(5);`
It just waited 5 seconds then gave the same result... 
```
*WM: WiFi save
*WM: Sent wifi save page
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Waiting for connection result with time out
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
*WM: Connection timed out
wifi evt: 7
*WM: Connection result: 
*WM: 0
*WM: Failed to connect.
*WM: Request redirected to captive portal
wifi evt: 7
```

tablatronix, 2018-08-22:
have you tried staging master latest ?

tablatronix, 2018-08-22:
cannot reproduce with wm master and esp8266 staging

tablatronix, 2018-08-22:
oh which lwip are you using ?

andysc, 2018-08-22:
not sure what you're asking - current release of wifimanager with not-released-yet version of ESP core? (sorry, I don't use GitHub very much)

I tried the dev branch of wifi manager with the current release of ESP core and that works fine, so when that's released, all should be good.

lwip is "v2 lower memory"

tablatronix, 2018-08-23:
Hmm I didnt see any changes to esp core since last release that would have fixed any issues. 
Ill test 2.4.2 also.

Maybe try lwip 1.4? There are some outstanding bugs with wifi disconnect/reconnect

andysc, 2018-08-23:
it's 2.4.2 core that's causing the problem with v0.14 of WiFiManager.

andysc, 2018-08-23:
ah, that's exactly what setConnectTimeout() does :) TVM

tablatronix, 2018-08-23:
I wonder if adding a disconnect before begin() in connectwifi would fix this hmm

tablatronix, 2018-08-23:
yeah thats why I suggested lwip.

I just tested this and could not reproduce master branch 2.4.2

andysc, 2018-08-23:
yay! that works :)
```WiFi.disconnect(false);```

tablatronix, 2018-08-23:
I suspected when it says idle_status it is between modes, or thinks sta is not on.

so disconnect then begin() fixes it ( this is implemented in development branch already )

andysc, 2018-08-23:
\o/ yay! That worked :)
```WiFi.disconnect(false);``` 
before the connect.

tablatronix, 2018-08-23:
pushed to hotfixes as test

tablatronix, 2018-08-23:
I have no idea what causes this, but this is the workaround, you can test hotfixes branch if you want. Not merging into master just yet, until I know whats going on

andysc, 2018-08-23:
didn't see your commit before I tried that.
Where is the WiFi_Disconnect() function defined? I get a compile error as it can't find that.

tablatronix, 2018-08-23:
oops
hmm there is already a disconnect in the code
 but only for existing creds

andysc, 2018-08-23:
where does ```wifi_station_disconnect``` live? Is that preferred?

andysc, 2018-08-23:
the UART stuff wrapped round it looks a bit like it's probably *not* preferred ;)

tablatronix, 2018-08-23:
thats how it is in WiFi.disconnect but I am not sure if its needed for the config saving or the disconnect

tablatronix, 2018-08-23:
If you use WiFi.disconnect you have to toggle persistent off and stuff or else it can erase your creds.
Not sure what you mean by preferred, I prefer it..

tablatronix, 2018-08-23:
Hey I added a debug status to hotfixes, if you can test that if you have time and let me know what the debug output says I would appreciate it.

andysc, 2018-08-23:
yup - the version in hot fixes works.

The two times that status gets printed, right at the start, and here when it reconnects, it's 0 both times...

```
*WM: Handle root
*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Status:
*WM: 0
*WM: Waiting for connection result with time out
*WM: Connection result: 
*WM: 3
connected to wifi!
```

andysc, 2018-08-23:
very odd - it's not working now. I think I had the wrong lwip version, but I've tried with 2 and 1.4 and neither work.

tablatronix, 2018-08-23:
you said you put a mode(sta) before maybe that is what fixed it ?

andysc, 2018-08-23:
OK ... so it doesn't work for me with your fix:
```
    ETS_UART_INTR_DISABLE();
    wifi_station_disconnect();
    ETS_UART_INTR_ENABLE();
```
but it DOES work with my earlier suggestion:
```
WiFi.disconnect(false);
```
(with lwip 2 and 2.4.2 core)

andysc, 2018-08-23:
I'll try putting the mode(STA) back in with your version... l8r

andysc, 2018-08-23:
putting in ```  WiFi.mode(WIFI_STA);``` before your debug status line only has the effect of closing the portal window.
It still doesn't connect.
I think we might have to go with my fix ;)

andysc, 2018-08-23:
ah, best to put in the mode change too - otherwise the portal window closes after the device reconnects...

andysc, 2018-08-23:
OK - actually the mode change doesn't affect the window closing.
so my change to your hot fix is:
```
    //trying to fix connection in progress hanging
/*
    ETS_UART_INTR_DISABLE();
    wifi_station_disconnect();
    ETS_UART_INTR_ENABLE();
*/
        // ASC
    WiFi.disconnect(false);
```

tablatronix, 2018-08-23:
hmm, it is literally the same code

```C++
bool ESP8266WiFiSTAClass::disconnect(bool wifioff) {
    bool ret;
    struct station_config conf;
    *conf.ssid = 0;
    *conf.password = 0;

    ETS_UART_INTR_DISABLE();
    if(WiFi._persistent) {
        wifi_station_set_config(&conf);
    } else {
        wifi_station_set_config_current(&conf);
    }
    ret = wifi_station_disconnect();
    ETS_UART_INTR_ENABLE();

    if(wifioff) {
        WiFi.enableSTA(false);
    }

    return ret;
}
```

tablatronix, 2018-08-23:
could it be a race condition ?
add delay(500) after it

andysc, 2018-08-23:
oh wow, so it is the same code - that's nuts. 
Yeah - could be race condition - I'll try that.

No, haven't tried either of those things, and no you didn't ask... good call - I'll try that too

andysc, 2018-08-23:
adding a delay between the disconnect code and the begin didn't help.

tablatronix, 2018-08-23:
Well what ever works, until someone else reproduces just use whatever you need to
ill try changing it later

andysc, 2018-08-23:
just had a thought - I'm configuring the **same** wifi credentials each time (I turn off my wifi hotspot so the ESP doesn't see it, it goes into config portal and then I turn the wifi on and then reconfigure the same wifi settings in WM).
So... in wifi.begin() it checks to see if the config is the same, and does something different (does a set config) if so.
In my case, it *is* the same (well, as far as SSID/password concerned) so maybe that's causing something?

I erased the wifi credentials and it worked fine. Then the next time it didn't (once I'd set some credentials).

andysc, 2018-08-23:
aha!
So, with the existing code, when it tries to reconnect it says: 
```
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Status:
*WM: 0
STA config unchanged
*WM: Waiting for connection result with time out
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
*WM: Connection timed out
*WM: Connection result: 
*WM: 0
*WM: Failed to connect.
```

If I go into 
`Library/Arduino15/packages/esp8266/hardware/esp8266/2.4.2/libraries/ESP8266WiFi/src`
and change `ESP8266WiFiSTA.cpp`

in function `begin(...)`
`   if (sta_config_equal(conf_compare, conf)) {`
to  
`  if (0)`
i.e. so even if the new credentials are the same, it still calls
`wifi_station_set_config_current`

Then it works! (with your hot fix version of `WiFiManager.cpp`).

The output this time is:
```
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Status:
*WM: 0
*WM: Waiting for connection result with time out
wifi evt: 2
scandone
switch to channel 6
state: 0 -> 2 (b0)
wifi evt: 7
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with PiWiFi, channel 6
dhcp client start...
wifi evt: 0
ip:172.20.10.3,mask:255.255.255.240,gw:172.20.10.1
wifi evt: 3
*WM: Connection result: 
*WM: 3
station: f4:5c:89:a1:54:c3 leave, AID = 1
rm 1
bcn 0
del if1
pm open,type:2 0
mode : sta(5c:cf:7f:wifi evt: 6
59:ff:62)
connected to wifi!
```

BTW I couldn't work out  how to get the output from that DEBUGV in there to say the STA config is the same, so I replaced it with a `Serial.println` and it displayed that (before I made the change!) when the wifi credentials were the same. So that confirmed it wasn't calling the set config when the credentials are the same.

andysc, 2018-08-23:
BTW, how do I turn off WM debug? It just seems to be on all the time

PProvost, 2018-08-29:
Any updates on resolving this with the current ESP Core, etc.? From the thread above, it isn't clear that the hotfix will resolve it. Thanks.

andysc, 2018-08-29:
my fix (forcing it to call `wifi_station_set_config_current`, whatever the wifi settings), is working fine for me.

tablatronix, 2018-08-29:
I know what the issue is I think.
Gonna check

tablatronix, 2018-08-29:
ok can you post exactly how to reproduce

- wm master
- esp 2.4.2
- save credentials in esp
- take ap down
- reset
- autoconnect fails -> cp starts
- enter cp , resave the existing saved credentials ( turn ap back on )
- click save
- ????

Is this correct?

tablatronix, 2018-08-29:
can you try
` WiFi.enableSTA(false);` right before the begin()

```
    //trying to fix connection in progress hanging
    ETS_UART_INTR_DISABLE();
    wifi_station_disconnect();
    ETS_UART_INTR_ENABLE();
    WiFi.begin(ssid.c_str(), pass.c_str());
```

tablatronix, 2018-08-29:
I added some test fix
forcing connect in begin
res = WiFi.begin(ssid.c_str(), pass.c_str(),0,NULL,true);
if(res !== WL_CONNECTED) DEBUG_WM(F("[ERROR]")+" "+(String)res);

PProvost, 2018-08-29:
@andysc  - above you mentioned that your fix (forcing it to set...) seems to be working. Did you make that fix against the current release or against the hotfix branch? It is unclear trying to follow the thread how to make the workaround. I don't suppose you have a fork repo with the change already in place? :)

andysc, 2018-08-30:
I agree it was a kludge ;)
But the WM code was the vanilla 0.14.0 master branch install.
The hack was to a file in the ESP wifi core code

andysc, 2018-08-30:
re: 
`>ok can you post exactly how to reproduce`
Yes, that's right
Just to be clear:
After the reconnect fails (because the AP is off), and it enters AP mode, 
`turn back on the AP`
Then when the CP window opens, your (previously configured) AP is visible.
`Select the previously configure AP, enter password`
`click save`

[20 more replies]

commit d6e091f5 (2018-08-29): test fix #702
closed 2020-01-22 by tablatronix