tzapu/WiFiManager #704: Try to reset memory when button is pressed

tomasbond, 2018-08-21:
### Basic Infos

#### Hardware
WiFimanager Branch/Release:  Development

Esp8266:  Nodemcu v3

Hardware:			ESP-12e

Core Version:      	2.4.0, staging

### Description

Trying to make the onboard flash run a reset on the wifi memory, running the wifiManager.resetSettings() function.

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp

#include <Arduino.h>
#define button 0
void setup() {
pinMode(button, INPUT);
}

void loop() {
if (digitalRead(button) == HIGH) {
Serial.println("Memory reset");
wifiManager.resetSettings();
}
```

### Debug Messages

Where should i put that code so when i press the flash button in the board it runs a memory reset and forget the ssid and password? tryied to put it here and doesnt work. On terminal i can see a New Client but nothing happens when the button is pressed

```
void loop(){
  WiFiClient client = server.available();   // Listen for incoming clients
  
if (digitalRead(button) == HIGH) {
Serial.println("Memory reset");
wifiManager.resetSettings();
}

  if (client) {                             // If a new client connects,
    Serial.println("New Client.");          // print a message out in the serial port
    String currentLine = "";                // make a String to hold incoming data from the client
    
    while (client.connected()) {            // loop while the client's connected
      
      if (client.available()) {             // if there's bytes to read from the client,

```

tablatronix, 2018-08-21:
I think you have to reboot after erasing settings.
I mean what do you wxpect to happen?

Do a printdiag and see that its empty

tablatronix, 2018-08-21:
did you set the pinmode in setup ?

tablatronix, 2018-08-21:
you have a while loop
while (client.connected()) {  
so yeah its gonna stay in there

tablatronix, 2018-08-29:
please close if sovlerd

emry78, 2019-04-14:
HI All
I'm emry beginner. i read all the  conversation about reset button. But i cannot emigine the reset connect to. i m using huzzah esp8266. i run autoconnect on wifimanager library. Need help to show the connection reset button and code in arduino ide. Thanks.

andysc, 2021-11-24:
I'm trying to make the ESP8266 forget its wifi credentials - I've tried 
`wifiManager.resetSettings();`
and
`Wifi.disconnect(true);`
with 
`delay(2000);`  
`ESP.reset();`
thrown in for good measure, but every time it comes back with the last successful connection details, saying:
`*WM: Using last saved values, should be faster`

I've even tried setting them to nonsense values (using `Wifi.begin("xxx", "xxx");` but it still goes back to the old ones on next reboot.

Is there a reliable way to nuke the credentials?

tablatronix, 2021-11-24:
That is supposed to work...

andysc, 2021-11-25:
oh, it looks like I'm hideously back-level! 
`0.15.0-beta` !!!
Sorry... I'll upgrade and report back

andysc, 2021-11-25:
should I be trying 2.00 or 0.16 ?? Are there breaking changes in 2.0 ?

andysc, 2021-11-25:
OK, here's my little test case: 
```
#include <ESP8266WiFi.h>

#include <WiFiManager.h>
WiFiManager wifiManager;

void setup() {

  Serial.begin(9600);
  delay(500);

  Serial.print("Looking for ");
  Serial.print(WiFi.SSID().c_str());
  Serial.print(" / ");
  Serial.println(WiFi.psk().c_str());

  Serial.println("*** Resetting WiFi credentials ***");
  
  wifiManager.resetSettings();
  
  delay(2000);
  ESP.reset();
}

void loop() {

}
````

When I run that I get...
```
⸮⸮⸮ԅp⸮@H6⸮0⸮⸮Looking for xxx / xxx
*** Resetting WiFi credentials ***
*WM: settings invalidated
*WM: THIS MAY CAUSE AP NOT TO START UP PROPERLY. YOU NEED TO COMMENT IT OUT AFTER ERASING THE DATA.
?)⸮)⸮⸮̍!H⸮Looking for xxx / xxx
*** Resetting WiFi credentials ***
*WM: settings invalidated
*WM: THIS MAY CAUSE AP NOT TO START UP PROPERLY. YOU NEED TO COMMENT IT OUT AFTER ERASING THE DATA.
⸮⸮⸮ԅ⸮⸮@H⸮⸮⸮⸮⸮Looking for xxx / xxx
```
so it the `resetSettings()` doesn't seem to have any effect .

This is on `v0.16.0`

andysc, 2021-11-25:
On `v2.0.5-beta`
... it works great!

```
** Resetting WiFi credentials ***
*wm:[1] resetSettings 
*wm:[1] SETTINGS ERASED 
H!⸮⸮L⸮⸮⸮J(⸮⸮⸮	⸮Looking for  / 
*** Resetting WiFi credentials ***
*wm:[1] resetSettings 
*wm:[1] SETTINGS ERASED 
H⸮⸮ԅ⸮⸮@H⸮⸮⸮⸮⸮Looking for  / 
*** Resetting WiFi credentials ***
*wm:[1] resetSettings 
*wm:[1] SETTINGS ERASED 
⸮⸮⸮⸮U9⸮@H⸮9⸮⸮⸮⸮Looking for  / 
```

OK - you can close this now :)

andysc, 2021-11-25:
aha - I see there's some additional magic in this version...
```
    WiFi.persistent(true);
    WiFi.disconnect(true);
    WiFi.persistent(false);
```
Thanks :)

andysc, 2021-11-25:
looks like there's more to it than that, as just adding those lines into my sketch doesn't erase the credentials.
Anyway... latest library seems to be the answer :)

andysc, 2021-11-25:
Cracked it :)
```
  WiFi.mode(WIFI_STA);
  
  WiFi.persistent(true);
  WiFi.disconnect(true);
  WiFi.persistent(false);
```
THAT resets the wifi credentials.
(you do not need to restart the ESP afterwards for it to take effect)

tablatronix, 2021-11-25:
Not sure what you mean , are you trying to fix this in v0?

andysc, 2021-11-25:
yes, I was trying to work out what you did in v2 that makes it work, that didn't work in v0.

tablatronix, 2021-11-26:
The whole thing was reworked, so no idea. Did not know it was not working. Ill look and maybe patch it maybe. But its totally worth switching to v2 at this time if you have the flash