tzapu/WiFiManager #706: reconnect to saved ap on power failure

tablatronix, 2018-08-22:
There is an issue with short lapses in connectivity and especially power failures that need a special handling to deal with either by the user or the library.

Ideally we would fire up the softap and leave sta on to connect, if status is connected we can abort the configportal automatically or after a timeout.

The problem with this is having a non connected sta makes softap unstable if not completely unusable. The theory is that the channel scanning autoconnect does makes the softap drop clients.

With this in mind there are a few options to resolve this.

- leave sta+ap mode and hope for the best...
- startap with wifi.channel to make the channel static, might increase stability.
- stop sta, start ap, do background scans for ap, if default ap is found in scan, restart sta snd try to connect, abort softap on result.
- if no softap clients switch to sta and try to reconnect back, rinse repeat
- use a long connecttimeout and a short configportaltimeout, on exit, if still not connected reboot
- let user do it themselves, do not use autoconnect, use user code to check and timeout after power failure determined via reset code and handle accordingly start/stop configportal as needed.

It depends on user case, but If there is a  saved credential, the odds of a power failure is higher than reconfigure of credentials. So waiting a minute for the router before starting configportal is a practical solution, also using an ondemand configportal removes this issue entirely.

Another options is to use some state tracking, if the last reboot was pwr, and last status was connected and time since last reboot is x, odds are router is rebooting...

tablatronix, 2018-08-23:
added several private options for testing this

```
    bool          _disableSTA             = false; // disable sta when starting ap, always
    bool          _disableSTAConn         = true;  // disable sta when starting ap, if sta is not connected ( stability )
    bool          _channelSync            = false; // use wifi channel when starting ap
```

aburst42, 2019-01-18:
I am encountering the same issue.

The current code seems to do something very close to the desired functionality, except it leaves the STA up while also serving the soft AP, which seems to interfere with the next attempt to run the chip in STA mode. Would love to see a solution to this that follows the description above:
"[when attempting manual resolution] stop sta, start ap, do background scans for ap, if default ap is found in scan, restart sta snd try to connect, abort softap on result."

In a nutshell, the desire is for an ReconnectWiFi function on the WifiManager, which attempts three steps, in sequence:
0. If (WiFi.status()==WL_CONNECTED), all is good - return true. Otherwise, proceed to 1
1. Attempt to connect with saved ssid/password, if available (AP is off, STA is on), until _connectTimeout expires or (WiFi.status()==WL_CONNECTED). If (WiFi.status()==WL_CONNECTED), all is good - return true. Otherwise, or if no saved ssid/password pair is available, proceed to 2.
2. Let user manually configure WiFi (AP is on, STA is off), until _configPortalTimeout expires or WiFi.status() is WL_CONNECTED. Return (WiFi.status()==WL_CONNECTED)

The user could then write this in their loop() function:
void loop() {
  if (wifiManager.ReconnectWiFi(AP_SSID, AP_PWD)) {
    // WiFi connection is ready for use.
  }
}

This would allow very robust self-healing (by repeating step 1) as well as manual healing (by repeating step 2) any time an interruption occurs, be it transient (power outage, AP reboot, etc.), or permanent (SSID change, password change, security mode change, etc.)

Happy to help test a solution through:
1. Simulated power cycles of the AP (disabling and re-enabling the wireless network)
2. SSID changes
3. Password changes
3. Security scheme changes.

Thanks!

Cheers,
~alex

tablatronix, 2019-01-18:
that is literally what I posted above...

aburst42, 2019-01-18:
I feel like I'm missing something... I wrote the trivially simple sketch below... I let the board connect to WiFi (with saved credentials), then I turned off the wireless network. Predictably, after a few seconds, the board lost connectivity and started serving the captive portal. I then turned the wireless network back on and, although the board continued to cycle between trying to connect to WiFi and serving the captive portal, it was never able to reconnect to WiFi. When I physically reset the board, it connected to WiFi in a couple of seconds.

What is your recommendation for changing the sketch below without: 1. Losing most of the benefits of using the WiFiManager class and 2. Repeatedly powercycling the board (can't be good for it over the long term...)

#include <ESP8266WiFi.h>

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>
#include <ArduinoOTA.h>

void setup() {
}

const unsigned long CONNECT_TIMEOUT = 180;         // Wait 3 minutes to connect to the real AP before trying to boot the local AP
const unsigned long AP_TIMEOUT =      180;         // Wait 3 minutes in the config portal before trying again the original WiFi creds
const char*         AP_SSID =         "ARDUINO";   // Create an access point with this SSID
const char*         AP_PWD =          "CONFIGURE"; // Protected the access point with this password

bool EnsureWiFi() {
  if (WiFi.status() != WL_CONNECTED) {
    WiFiManager wifiManager;

    wifiManager.setConnectTimeout(CONNECT_TIMEOUT);
    wifiManager.setTimeout(AP_TIMEOUT);

    wifiManager.autoConnect(AP_SSID, AP_PWD);
  }

  return (WiFi.status() == WL_CONNECTED);
}

void loop() {
  if (EnsureWiFi()) {
    // WiFi connection is ready
  }
}

tablatronix, 2019-01-18:
what branch are you using ?

aburst42, 2019-01-18:
The one that the Arduino Studio library manager installs automatically (version appears to be 0.14.0) (see attached screenshot). Should I be using a different, manually installed one?
[image]

aburst42, 2019-01-18:
Also... If scanning for APs is what's causing the issues, I would gladly trade that (i.e. type the ssid/password in a text box, rather than selecting it) in order to have rock solid auto-recovery when possible.

tablatronix, 2019-01-18:
Yeah no support or development is being done on anything but development branch

aburst42, 2019-01-18:
Got it. I'll clone from https://github.com/tzapu/WiFiManager/tree/development and report back with the results. Thanks!

tablatronix, 2019-01-18:
hopefully this is a known issue and already fixed, but let me know

aburst42, 2019-01-20:
I am very happy to report that it worked exactly as intended. Also, I noticed that the behavior that enables this library to work as expected is much more sophisticated than it was on the master branch - thank you for improving the code and making it work for my scenario!

tablatronix, 2019-01-20:
yeah it still needs to be documented and examples cleaned up, that is mostly why it is not beta yet

alvaroaguero55, 2019-08-11:
Still same issue with latest versión?

tablatronix, 2019-08-11:
yes there has not been a release since this, use development branch

alvaroaguero55, 2019-08-11:
Which is the link of the working development branch?

Bolukan, 2019-10-07:
Suggestion to the opening post: If the server gets a client request via the AP, disconnect STA for some time (so AP-only mode). Maybe like a watchdog counter (each server-request reset time counter). Maybe even a client connection (before the page-request) may activate this behaviour.

tablatronix, 2019-10-07:
@Bolukan it is worth testing, But the main problem is the ap can be entirely unstable when sta is trying to connect, but it would be nice to find a stable solution to background connecting, ideally users can just use non blocking and make their own logic, this can probably be POC in user code atm.

Bolukan, 2019-11-16:
I am aware of the lack of progress. It was quiet timeconsuming (wiping, downloading and analysuing flashes) but bottom-line too difficult for me to create reproducable testcases.

tablatronix, 2019-11-16:
This is not something I will look into until next version

Jhonathaned, 2020-01-22:
Any progress on this? I have the very same issue reported here #738.

I want to use "AutoConnect" on setup, so It will run once on boot, so lets suppose that It failed to connect to the saved credentials, and that the ConfigPortalTimeout was hit: Now I need It trying to reconnect to saved credentials forever, but It doesn't do that, 'cause when I start up my router/AP, it simply doesn't connect.

When disconnected, I cannot call the "AutoConnect" function again in the loop because it will enter on AP mode, and the really problem of it is that it stucks the code, as it is using an "while(1)" inside library. The code being stuck is a real problem for me, as I am using the NRF24Mesh library, which needs to have its "mesh.update()" function cycled in loop, otherwise it will not keep my mesh network up.

I'm using the latest library version provided by the Arduino IDE's library manager.

I would really appreciate any kind of help.

ftaibi, 2020-09-08:
Hi @aburst42 

I have tried your [solution](https://github.com/tzapu/WiFiManager/issues/706#issuecomment-455632410) and does not work for me. Any idea why?

After losing the wifi connection switch to AP but I turn the wifi on again and does not reconnect. 

Cheers

alexburs, 2020-09-11:
Hi @ftaibi,

Unfortunately, I don't know why the solution wouldn't work for you. I included below the full code of my garage controller. Beyond staying connected to WiFi through power outages and such, it doesn't do much - it just sets a pin to activate the garage door. Maybe give it a go on your Arduino with a copy of WifiManager from the development branch?

`#include <ESP8266WiFi.h>

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>
#include <ArduinoOTA.h>
#include <Ticker.h>

#define PRINTF_BUF 512 // define the tmp buffer size (change if desired)

// For control
ESP8266WebServer server(80);

//DEBUG
// Change this to true to collects logs
const bool DEBUG_MODE = true;
const int MAX_DEBUG_ENTRIES = 100;
long debugCodes[MAX_DEBUG_ENTRIES][2];
long crtDebugIndex = 0;

void logCode(long code)
{
  if (DEBUG_MODE)
  {
    debugCodes[crtDebugIndex][0] = code;
    debugCodes[crtDebugIndex][1] = os_getCurrentTimeSec();

    crtDebugIndex++;

    if (crtDebugIndex >= MAX_DEBUG_ENTRIES)
    {
      crtDebugIndex = 0;
    }
  }
}

void handleDebugInfo() {
  char szTmp[MAX_DEBUG_ENTRIES * 10];

  if (DEBUG_MODE)
  {
    unsigned long days = 0, hours = 0, minutes = 0;

    int length = 0;
    for (int crtDebugInfoIndex = 0;crtDebugInfoIndex < crtDebugIndex; crtDebugInfoIndex++)
    {
      long code = debugCodes[crtDebugInfoIndex][0];
      long val = debugCodes[crtDebugInfoIndex][1];
      days = val / (3600*24);
      val -= days * (3600*24);
      hours = val / 3600;
      val -= hours * 3600;
      minutes = val / 60;
      val -= minutes*60;
      
      length += sprintf(
                  szTmp + length,
                  "%02d %02d:%02d:%02d.%02d<br>",
                  (int)code,
                  (int)days, (int)hours, (int)minutes, (int)val);

    }
  }
  else // !DEBUG_MODE
  {
[... 202 lines cut ...]

  if (WiFi.status() == WL_CONNECTED) {
    ledOn();
  }

}

unsigned long os_getCurrentTimeSec()
{
  static unsigned int wrapCnt = 0;
  static unsigned long lastVal = 0;
  unsigned long currentVal = millis();

  if(currentVal < lastVal)
  {
    wrapCnt++;
  }

  unsigned long seconds = currentVal/1000;
  
  //millis will wrap each 50 days, as we are interested only in seconds, let's keep the wrap counter
  return (wrapCnt*4294967) + seconds;
}

`