tzapu/WiFiManager #707: default ap name does not match chip id

tablatronix, 2018-08-23:
I was testing the other day and I noticed that the default ap did not match the generated wm one from chip id. It was a bit odd.

try to reproduce , I have a screenshot somewhere

Kadah, 2018-08-24:
Do you mean this?
https://i.imgur.com/T7yBhBA.png

Testing using OnDemandConfigPortal.ino. Screenshot was of an ESP8266 (ESP-12F) after rebooting, but I got the same but different SSID on an ESP32 Thing.

tablatronix, 2018-08-24:
Yeah the one the chip generates it different that the one wm generates when no ssid is provided

tablatronix, 2018-08-24:
Will have to test esp32 and see if this fix covers that as well

tablatronix, 2018-08-29:
esp32

```
  // AP esp_267751
  // ESP32 Chip ID = 507726A4AE30
```

tablatronix, 2020-05-11:
- [ ] fix esp chipid

commit 1bf3fc04 (2018-08-24): fixes #707
closed 2021-01-04 by tablatronix