tzapu/WiFiManager #708: blank wifi credentials aren't handled correctly in connectWifi

andysc, 2018-08-23:
If you erase the wifi credentials and start WiFiManager, it says
```
Using last saved values, should be faster
```
whereas it should say
```
No saved credentials
```
and go straight into config portal mode.

It looks like 
```    if (WiFi.SSID()) {```
isn't correctly evaluating an empty SSID.

tablatronix, 2018-08-29:
You didnt fill out any information

andysc, 2018-08-29:
sorry... WM 0.14.0
ESP core 2.4.2
running on a  Wemos ESP8266

line 294 of WiFiManager.cpp
erase wifi credentials using the option in Arduino IDE (Erase sketch and WiFi settings)

(I found it while we were playing with #702, but I thought I'd raise it separately :))

tablatronix, 2018-08-29:
master branch ?

andysc, 2018-08-30:
yes

commit 8f9edd18 (2019-08-30): fixes #708
closed 2019-08-30 by tablatronix