tzapu/WiFiManager #709: wifimanager timeout implementation at the sketch

mikekgr, 2018-08-25:
Dear Sirs, I am using wifimanager development tree at my ESP8266 Arduino Core 2.4.2.
I need to set up a time out in the sense, when the hard coded SSID, PASSWORD doesn't achieve WiFi connection then to be activated the wifimanager in order to input using its web page, the new after scan PASSWORD for the known SSID if any. My requirement is in case that nothing is entered to wifimanager web page, to have specified a timeout (ex 10 minutes ) and if this reached then to return back to normal sketch execution. Looking at the examples I saw there two commands but I don't know how to use them:
wm.setConfigPortalTimeout(600);
wm.setConnectTimeout(10);

Please help me.

Thanks

tablatronix, 2018-08-26:
Configportaltimeout will do what you want

mikekgr, 2018-08-26:
Dear @tablatronix ,
thanks a lot for your fast reply. So, using setConfigPortalTimeout will do what I want! Could you also explain what setConnectTimeout does?
Many thanks.

tablatronix, 2018-08-26:
https://github.com/tzapu/WiFiManager/blob/development/WiFiManager.h#L169

mikekgr, 2018-08-26:
Dear @tablatronix 
Thanks. I found the Advanced.ino in the examples, that is an excellent sketch to see and try many thinks.
I truly appreciate your time and effort in this near perfect library.

Thanks and Best Regards,
Mike Kranidis

closed 2018-08-29 by tablatronix

jorgele, 2020-08-17:
Amigos, tengo un problema combinando el parametro "setConfigPortalBlocking(false);" y "setConfigPortalTimeout(xx)". Si configuro el primero para que el portal no bloquee el programa, no funciona el timeout.
¿Es así, o estoy haciendo algo mal?

tablatronix, 2020-08-18:
This is not a forum post new issues, do not post to old issues