tzapu/WiFiManager #721: softAPConfig failed

rodri16, 2018-09-05:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [x ] Development

**Esp8266/Esp32:**
- [x ] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
SDK Version 2.2.1
Core version 2.4.1
Boot version 4

### Description

I am using autoconnect and callbacks, to connect to a stored WiFi SSID. In callback I am putting static IP 10.0.1.1 for the config portal.
 myWiFiManager->setAPStaticIPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0)); 

If Wifi does not connect because I do a resetSettings(), config portal starts ok on 10.0.1.1.
When WiFi is connected using autoconnect and press a button which runs startConfigPortal() it goes to 192.168.4.1. Even adding setAPStaticIPConfig before startConfigPortal()

### Settings in IDE

Module:	Generic ESP8266 module
Flash size: 4M (3M SPIFFS)
all other parameters default
Additional libraries: 
ESP8266mDNS.h
TimeLib.h
NtpClientLib.h
DNSServer.h
HLW8012.h
ArduinoJson.h

### Debug Messages
Doing a wifiManager.resetSettings(); It works ok
*WM: [1] SETTINGS ERASED
*WM: [3] WiFi station enable
*WM: [3] enableSTA PERSISTENT ON
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: server
*WM: [2] Added Parameter: port
*WM: [2] Added Parameter: TimeZone
*WM: [1] AutoConnect
*WM: [2] Connecting as wifi client...
*WM: [3] WIFI station disconnect
*WM: [1] No saved credentials, skipping wifi
*WM: [2] Connection result: WL_NO_SSID_AVAIL
*WM: [3] lastconxresult: WL_NO_SSID_AVAIL
*WM: [1] AutoConnect: FAILED
*WM: [2] AccessPoint set password is VALID
*WM: [1] xxxx
*WM: [3] WIFI station disconnect
*WM: [3] WiFi station enable
*WM: [2] Disabling STA
*WM: [2] Enabling AP
*WM: [1] StartAP with SSID: xxxx
*WM: [1] Custom AP IP/GW/Subnet:
*WM: [1] AP IP address: 10.0.1.1
*WM: [3] setupConfigPortal
*WM: [1] Starting Web Portal
*WM: [3] dns server started with ip: 10.0.1.1
*WM: [2] HTTP server started
*WM: [2] WiFi Scan ASYNC started
*WM: [2] Config Portal Running, blocking, waiting for clients...
*WM: [2] WiFi Scan ASYNC completed in 2186 ms
*WM: [2] WiFi Scan ASYNC found: 9

And then pressing the button to start config portal
*WM: [2] AccessPoint set password is VALID
*WM: [1] xxxx
*WM: [3] WiFi station enable
*WM: [2] Enabling AP
*WM: [1] StartAP with SSID: xxxx
*WM: [1] Custom AP IP/GW/Subnet:
*WM: [0] [ERROR] softAPConfig failed!
*WM: [1] AP IP address: 10.0.1.1
*WM: [3] setupConfigPortal
*WM: [1] Starting Web Portal
*WM: [3] dns server started with ip: 10.0.1.1
*WM: [2] HTTP server started
*WM: [2] WiFi Scan ASYNC started
*WM: [2] Config Portal Running, blocking, waiting for clients...
*WM: [2] Portal Timeout In 118 seconds
*WM: [2] WiFi Scan ASYNC completed in 1589 ms
*WM: [2] WiFi Scan ASYNC found: 8

THEN aftera reboot with Wifi settings and press the button
*WM: [2] AccessPoint set password is VALID
*WM: [1] XXXXX
*WM: [3] WiFi station enable
*WM: [2] Enabling AP
*WM: [1] StartAP with SSID: XXXXX
Entered config mode
0.0.0.0
XXXXX
*WM: [1] Custom AP IP/GW/Subnet:
*WM: [0] [ERROR] softAPConfig failed!
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal
*WM: [1] Starting Web Portal
*WM: [3] dns server started with ip: 192.168.4.1
*WM: [2] HTTP server started
*WM: [2] WiFi Scan ASYNC started
*WM: [2] Config Portal Running, blocking, waiting for clients...
*WM: [2] Portal Timeout In 118 seconds
*WM: [2] WiFi Scan ASYNC completed in 1589 ms
*WM: [2] WiFi Scan ASYNC found: 5
*WM: [2] Portal Timeout In 88 seconds

Thanks

tablatronix, 2018-09-05:
Do you have sketch ?
Is WM global or are you setting apconfig in loop where button is pressed?

tablatronix, 2018-09-05:
I am just not seeing this at all

```
*WM: [2] Connection result: WL_CONNECTED
*WM: [3] lastconxresult: WL_CONNECTED
*WM: [1] AutoConnect: SUCCESS 
*WM: [1] STA IP Address: 10.80.30.158
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
wifi evt: 7
BUTTON PRESSED
*WM: [3] WiFi station enable 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  OnDemandAP
*WM: [1] Custom AP IP/GW/Subnet: 
[APConfig] local_ip: 10.0.1.99 gateway: 10.0.1.1 subnet: 255.255.255.0
[APConfig] DHCP IP start: 10.0.1.198
[APConfig] DHCP IP end: 10.0.1.42
*WM: [2] AP has anonymous access! 
bcn 0
del if1
pm open,type:2 0
add if1
pm close 7
dhcp server start:(ip:10.0.1.99,mask:255.255.255.0,gw:10.0.1.1)
bcn 100
*WM: [1] AP IP address: 10.0.1.99
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  10.0.1.99
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] Config Portal Running, blocking, waiting for clients... 
wifi evt: 7
scandone
*WM: [2] WiFi Scan ASYNC completed in 2198 ms
*WM: [2] WiFi Scan ASYNC found: 9
```

tablatronix, 2018-09-05:
you need to enable esp debugging wifi

rodri16, 2018-09-05:
WM is global
*WM: [2] AccessPoint set password is VALID 
*WM: [1] Dispositive 
*WM: [3] WiFi station enable 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  Disp_3
Entered config mode
0.0.0.0
Disp_3
*WM: [1] Custom AP IP/GW/Subnet: 
*WM: [0] [ERROR] softAPConfig failed! 
mode : sta(ec:fa:bc:00:a2:5f) + softAP(ee:fa:bc:00:a2:5f)
add if1
pm close 7
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
bcn 0
del if1
pm open,type:2 0
add if1
pm close 7
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] Config Portal Running, blocking, waiting for clients... 
*WM: [2] Portal Timeout In 118 seconds
scandone
*WM: [2] WiFi Scan ASYNC completed in 1599 ms
*WM: [2] WiFi Scan ASYNC found: 10

Sketch

```
#include <FS.h>                   //this needs to be first, or it all crashes and burns...
#include <ESP8266WiFi.h>          //https://github.com/esp8266/Arduino
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager
#include <DNSServer.h>
#include <TimeLib.h>
#include <NtpClientLib.h>
#include <ArduinoJson.h>          //https://github.com/bblanchon/ArduinoJson

// Use WiFiClient class to create TCP connections
WiFiClient client;

//for LED status
#include <Ticker.h>
Ticker ticker;

//-------------------------------------variables y definiciones a cambiar----------------------------------
#define nro_dis 3    //cambiar por el numero de dispositivo 
//----------------------------------------no cambiar-------------------------------------------------------
#define BUILTIN_LED 15 // green led 15, red+relay 12. LOW is on
#define RELAY_PIN 12 //relay

#define SSID1 "Disp_"
#define password "Dispositive"

String datenow, timenow, datenowdot, timenowdot;
unsigned long previousMillis = 0;
const long interval = 5*60*1000; //envio cada 5 minutos
volatile boolean butt_flag = false;

int8_t minutesTimeZone = 0;
[... 387 lines cut ...]

  server->on("/", handleRoot);

//---------------------mdns---------------
  if (!MDNS.begin(Buf)) {
    Serial.println("Error setting up MDNS responder!");
    while(1) { 
      delay(1000);
    }
  }
  Serial.println("mDNS responder started");
//---------------------webserver-------  
  server->onNotFound(handleNotFound);
  server->begin();
  Serial.println("HTTP server started");
  
//---- Add service to MDNS-SD
  MDNS.addService("http", "tcp", 80);
  butt_flag = false;
  }
  else {
  }
}

```

rodri16, 2018-09-05:
*WM: [2] AccessPoint set password is VALID 
*WM: [1] Dispositive 
*WM: [3] WiFi station enable 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  Disp_3
Entered config mode
0.0.0.0
Disp_3
*WM: [1] Custom AP IP/GW/Subnet: 
[APConfig] local_ip: 10.0.1.1 gateway: 10.0.1.1 subnet: 255.255.255.0
[APConfig] wifi_softap_dhcps_stop failed!
[APConfig] wifi_set_ip_info failed!
[APConfig] DHCP IP start: 10.0.1.100
[APConfig] DHCP IP end: 10.0.1.200
[APConfig] wifi_set_ip_info failed!
[APConfig] wifi_softap_set_dhcps_lease_time failed!
[APConfig] wifi_softap_dhcps_start failed!
[APConfig] IP config Invalid?!
*WM: [0] [ERROR] softAPConfig failed! 
mode : sta(ec:fa:bc:00:a2:5f) + softAP(ee:fa:bc:00:a2:5f)
add if1
pm close 7
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
wifi evt: 8
wifi evt: 8
bcn 0
del if1
pm open,type:2 0
add if1
pm close 7
dhcp server start:(ip:192.168.4.1,mask:255.255.255.0,gw:192.168.4.1)
bcn 100
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] Config Portal Running, blocking, waiting for clients... 
*WM: [2] Portal Timeout In 118 seconds
scandone
*WM: [2] WiFi Scan ASYNC completed in 1600 ms
*WM: [2] WiFi Scan ASYNC found: 12
wifi evt: 7

tablatronix, 2018-09-05:
cant reproduce, it has something to do with your other code, or the mode its in

`wifi_softap_dhcps_stop failed`
I have no idea why this would be failing, maybe conflict with starting mode, or is not switching ap correctly.

It could also be an open issue I have in esp with enableAP

tablatronix, 2018-09-05:
I also wonder if swapping the order of these 2 would help

```
// wifimanager.cpp
  // setup optional soft AP static ip config
  if (_ap_static_ip) {
    DEBUG_WM(F("Custom AP IP/GW/Subnet:"));
    if(!WiFi.softAPConfig(_ap_static_ip, _ap_static_gw, _ap_static_sn)){
      DEBUG_WM(DEBUG_ERROR,"[ERROR] softAPConfig failed!");
    }
  }

  #ifdef ESP8266
    // @bug workaround for bug #4372 https://github.com/esp8266/Arduino/issues/4372
    if(!WiFi.enableAP(true)) {
      DEBUG_WM(DEBUG_ERROR,"[ERROR] enableAP failed!");
      return false;
    }
    delay(500); // workaround delay
  #endif
```

and place the enableap before the softAPConfig

rodri16, 2018-09-06:
Yes you are right @tablatronix! Swapping enableAP before softAPConfig problem solved!!! Thank you!

tablatronix, 2018-09-20:
#722

tablatronix, 2018-10-12:
I moved the wifiManager.setAPCallback(); after startap, I think will break this implementation

tablatronix, 2020-05-11:
Where did we leave off here ? ughh

I think its safe to close, might need to reopen a new issue for more callbacks if needed

closed 2021-06-21 by tablatronix