tzapu/WiFiManager #723: Feature request: async wifi config

joba-1, 2018-09-06:
Hi,

I used your lib successfully to configure wlan for a small project, very nice stuff, thanks.

Now I would like to use it for something else, where wifi is not necessary, just nice to have.

This means: I don't want the portal or wifi setup to block setup() or loop(), just setup wifi in setup() if config is available, else, if a button is pressed (or whatever other trigger) serve the wifi configuration portal **in the background** for some time, **while the progam continues to do its usual job** (in my case starting a fan if humidity is too high).

I have not fully understood the architecture of your lib, maybe this is already possible somehow, but if not: how about basically cutting out the while(1) loop out of startConfigPortal() as bool WiFiManager.handlePortal() where it returns false instead of yield() and true instead of break. Then I could call startConfigPortal() on request or in setup() and call handlePortal() in loop() without disturbing the more important stuff going on there.

tablatronix, 2018-09-06:
development branch

closed 2018-09-06 by tablatronix

joba-1, 2018-09-06:
How?
I changed to dev branch, then opened an issue. Not enough?

tablatronix, 2018-09-06:
There are several issues discussing this, this is not a forum for support, its for reporting issues.
Did you even search?

joba-1, 2018-09-06:
just forget about it. I‘ll implement it for myself if I need it.

-- 
iphone so i typo

tablatronix, 2018-09-07:
https://github.com/tzapu/WiFiManager/issues/379

tablatronix, 2018-09-07:
https://github.com/tzapu/WiFiManager/blob/development/examples/AutoConnect/AutoConnectNonBlocking/AutoConnectNonBlocking.ino