tzapu/WiFiManager #724: WifiManager not working with code having EEPROM and PUBSUB library added

amguamgu, 2018-09-11:
### Basic Infos

I am using basic sketch to on/off the relay. I have used EERPROM.h to save the last state of relay and also PubSubClient.h library to drive the circuit using MQTT.

I have addded WifManager library to manage the credentials of wifi only. 
But code is not working.. some issue is there no debug is also coming...

#### Hardware

ESP-01, relay module

**WiFimanager Branch/Release:**
- [ ] Master 

**Esp8266/Esp32:**
- [ ] ESP8266

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01

**ESP Core Version:	2.4.0, staging**
- [ ] 2.4.0

### Description

Basically i am trying to use your library to manage wifi credentials. My device is working on MQTT protocol and also saving device state using EEPROM library

Please help as WifiManager is not working 

### Settings in IDE

Module:  			NodeMcu, Wemos D1

EERPROM.h, PubSubClient.h

### Sketch

```cpp

/*
    Created and managed by Amit Gupta
    Will use ESP8266-01 as the wifi module
    MQTT has been used as communication protocol
    Plug and Play device code 
    WiFiManage has also been used to configure connectivity to wifi

*/ 

#include <EEPROM.h>
#include <ESP8266WiFi.h>
#include <PubSubClient.h>
#include <ESP8266Ping.h>

#include <DNSServer.h>            //Local DNS Server used for redirecting all requests to the configuration    
#include <ESP8266WebServer.h>     //Local WebServer used to serve the configuration portal
#include <WiFiManager.h>          //https://github.com/tzapu/WiFiManager WiFi Configuration Magic

// write these with values suitable for your network.

#define GPIO0 0
#define GPIO2 2

const char* ssid  = "XXXXXXXX";
const char* password  = "XXXXXXX";

int gpio0_add = 0;
int gpio2_add = 1;

const char* mqtt_server = "XXXXXXXXXX"; 
const char* mqtt_username = "XXXXXXX";
const char* mqtt_password = "XXXXXXXX";

IPAddress ip(10, 0, 4, 17); // where xx is the desired IP Address
IPAddress gateway(10, 0, 4, 1); // set gateway to match your network
//Serial.print(F("Setting static ip to : "));
//Serial.println(ip);
IPAddress subnet(255, 255, 255, 0); // set subnet mask to match your

IPAddress dns1(4, 2, 2, 2); // set subnet mask to match your
IPAddress dns2(8, 8, 8, 8); // set subnet mask to match your

WiFiClient espClient;
PubSubClient client(espClient);
long lastMsg = 0;
char msg[50];
int value = 0;

// Will be called only for first time ...
void resetEEPROM(){
      EEPROM.write(gpio0_add, LOW);
      EEPROM.write(gpio2_add, LOW);     
}

// Method to connect to internet for called in the setup 
void setup_wifi() {

  delay(10);
  // We start by connecting to a WiFi network
  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  //comment this line if gateway and ip is not reqired
  WiFi.config(ip, dns1, gateway, subnet);
  
  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

  randomSeed(micros());

  Serial.println("");
  Serial.println("WiFi connected");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());
  delay(2000);

  //code to check whether connected network has internet or not
  bool ret = Ping.ping("www.google.com");
  if(ret == true){
     Serial.println("google");
     digitalWrite(GPIO0, LOW); 
  }else{
    Serial.println(" not google");
     //digitalWrite(ISCONNECTED, HIGH); 
  }
}

//Method will called some message is recived in the topic to take any action
void callback(char* topic, byte* payload, unsigned int length) {
 
  Serial.print("Message arrived [");
  Serial.print(topic);
  Serial.print("] ");
  for (int i = 0; i < length; i++) {
    Serial.print((char)payload[i]);
  }
  Serial.println();

  char tt[10];
  strcpy(tt,topic);
  
 Serial.println(tt);
 if(!strcmp(tt,"PP_ON_OFF")){
  
       // Switch on the LED if an 1 was received as first characte
       Serial.println("1");
        if ((char)payload[0] == '1') {
          EEPROM.write(gpio2_add,LOW);
          digitalWrite(GPIO2, LOW);   // Turn the LED on (Note that LOW is the voltage level
          // but actually the LED is on; this is because
          // it is acive low on the ESP-01) 
        } else {
           EEPROM.write(gpio2_add,HIGH);
          digitalWrite(GPIO2, HIGH);  // Turn the LED off by making the voltage HIGH
        }
        EEPROM.commit();
        
}else{
   Serial.println("Un- Known");
}
       
}

// Method will be called when MQTT protocal need to do connection/re-connection to MQTT Broker
void reconnect() {
  digitalWrite(GPIO0, HIGH); 
  // Loop until we're reconnected
  while (!client.connected()) {
    Serial.print("Attempting MQTT connection...");
    // Create a random client ID
    String clientId = "ESP8266Client-";
    clientId += String(random(0xffff), HEX);
    // Attempt to connect
    if (client.connect(clientId.c_str(), mqtt_username, mqtt_password)) {
      Serial.println("connected");
     
      // Once connected, publish an announcement...
      //--------------  client.publish("outTopic", "hello world");
      // ... and resubscribe
      client.subscribe("PP_ON_OFF");

      digitalWrite(GPIO0, LOW); 
    } else {
      Serial.print("failed, rc=");
      Serial.print(client.state());
      Serial.println(" try again in 5 seconds");
      // Wait 5 seconds before retrying
      
      delay(5000);
    }
  }
}

// Mehotd called when ESP8266 is in config mode
void configModeCallback (WiFiManager *myWiFiManager) {
  Serial.println("Entered config mode");
  Serial.println(WiFi.softAPIP());

  Serial.println(myWiFiManager->getConfigPortalSSID());
}

//============== SETUP METHOD ===============================

void setup() {
   EEPROM.begin(512);
   
   WiFiManager wifiManager;
   
    wifiManager.setMinimumSignalQuality(10);
    wifiManager.setRemoveDuplicateAPs(false);
    wifiManager.setAPCallback(configModeCallback);

if (!wifiManager.autoConnect("WIFI_DEVICE", "password")) {
    Serial.println("failed to connect and hit timeout");
    delay(3000);
    //reset and try again, or maybe put it to deep sleep
    ESP.reset();
    delay(5000);
}
  
  // resetEEPROM();
   
  pinMode(GPIO0, OUTPUT); // Initialize the BUILTIN_LED pin as an output
  pinMode(GPIO2, OUTPUT); 
  
  digitalWrite(GPIO2, EEPROM.read(gpio2_add)); 
  Serial.begin(115200);

  setup_wifi();
    
  client.setServer(mqtt_server, 1883);
  client.setCallback(callback);
  
}

//========= SETUP METHOD END =============================

//=========== LOOP METHOD=============================
// Main loop method ESP8266-01
void loop() {

  if (!client.connected()) {
    reconnect();
    //digitalWrite(ISCONNECTED, HIGH); 
  }
  client.loop();

  long now = millis();
  if (now - lastMsg > 3000) {
    lastMsg = now;
    

 int  gpio0_status = digitalRead(GPIO0);
 int  gpio2_status = digitalRead(GPIO2);

   Serial.println("--currst---");

String currst = "";
  currst.concat(gpio0_status);
  currst.concat(gpio2_status);

   
   char const* finalstatus = currst.c_str();
     client.publish("PP_ON_OFF_STATUS", finalstatus);
    return;
  }
}
//===== LOOP METHOD END================

```

### Debug Messages

No debug messages are coming on serial montor

tablatronix, 2018-09-11:
your sketch does not have a serial begin

tablatronix, 2018-09-12:
you need to put it earlier

tablatronix, 2018-10-12:
can this be closed?

tablatronix, 2019-01-23:
I hate libraries that implement their own connection handling

tablatronix, 2019-01-23:
This is strange , have you tried giving dns a delay to start? Before pubsub connect, delay 2 seconds? I am wondering why hostname resolution would be failing, there are some other issues that have been reported similarly very odd

0x2f0713, 2020-04-26:
> ```c++
>     IPAddress ipAddr;
>     if (WiFi.hostByName(get_mqttServer().c_str(), ipAddr) != 1)
>         Serial.println("ERROR: Unable to resolve host name!");
>     else
>         Serial.printf("INFO: Host name resolved successfuly - %s\n", ipAddr.toString().c_str());
> 
>     // snipped
> 
>     mqtt_client.setServer(ipAddr, 1883);
> ```

Many thanks, this code worked for me!