tzapu/WiFiManager #725: Time to Connect

josemariaaraujo, 2018-09-11:
(ESP Core Version:	2.4.1)
Hi, 
I've trying to get my power consumption down in a battery powered sensor, and after some profiling I've found that most of my awake time was spent in WiFiManager autoConnect (with a static IP) , spending always more than 3 seconds to establish a connection, much longer than if done directly with `WiFi.config(..); WiFi.begin(); while ( WiFi.status() != WL_CONNECTED ) delay(1);` which only took 0.2-0.3s.

After going deeper in your code, I've found the culprit: in `int WiFiManager::connectWifi` if I comment  `      ETS_UART_INTR_DISABLE();
      wifi_station_disconnect();
      ETS_UART_INTR_ENABLE();` I get the same connection times as using  the code above.

I don't know enough of the ESP8266 to understand if what I have commented is still relevant and should be left alone. But maybe someone is also trying to get the lowest consumption while still using WiFiManager for it's benefits, and this issue/solution will help them.

Thank you for your great work!

tablatronix, 2018-09-12:
Yeah its already detsiled in development branch

tablatronix, 2018-09-12:
#598

closed 2018-09-12 by tablatronix