tzapu/WiFiManager #728: Power saving with WiFi Manager

grgoncal, 2018-09-12:
Hi. I'm working on a project with batteries and I'm using WiFi Manager. 

Is there a way to enter in light sleep mode for a while and then reconnect with WiFi Manager? I've tried this, but it does not reconnect by itself.

```
    Serial.println("Sleeping...");
    WiFi.disconnect();
    WiFi.mode(WIFI_OFF);
    WiFi.forceSleepBegin(5 * 1000000L); //In uS. Must be same length as your delay
    delay(5 * 1000); //Hang out at 15mA for 6 seconds
    WiFi.mode(WIFI_STA);
```

tablatronix, 2018-09-12:
No idea, never used it

Here is the mega issue
https://github.com/esp8266/Arduino/issues/1381

grgoncal, 2018-09-12:
Links2004 Response to the tablatronix link worked for me. 

Thank you!

closed 2018-09-12 by grgoncal