tzapu/WiFiManager #730: Adding images?

arrowcircle, 2018-09-14:
Hello!
I want to add svg logo to my template of WiFiManager.
I tried to add it inline as base64, but bottom of the page is cut.
I think its because 30kb svg image added to web page.

I tried to add separate route to the image and serve it via progmem, but now sometimes I got broken image. As I understand its because of blocking nature of webserver inside wifimanager, that does not support cuncurrent requests and dont allow waiting.

What is the best way to add 30kb logo to the template?

tablatronix, 2018-09-14:
use development branch and post example

diegoalbaga, 2018-09-18:
How do I use the development branch?, I have tried to add an image too and it work at the html template but it wont let me compile at arduino ide