tzapu/WiFiManager #731: Disable deleting exiting credentials when starting wifiManager

henkep, 2018-09-16:
I have existing credentials saved on my device, I then push a button on my device to manually start the wifiManager. By default the manger then deletes my existing credentials, I would like it to remember credentials until new credentials are being entered. A real life scenario is, what if someone by misstake pushes the button on the device, I have the timeout set to 3 min. If no credentials are saved after 3 min the wifiManager shuts down but the ESP has lost the esisting credentials and need to be reconfigured.

Is there anyway for the manager to keep the existing credentials until new credentials are provided?

henkep, 2018-09-16:
Hi
Tested develop and it does the same thing.

henkep, 2018-09-17:
OK so this was my bad, I did a WiFi.disconnect() before starting the wifiManager and that messed with the settings. But when I removed that I get another problem. So the scenario is this, the user is connected to SSID 1 and everything is working just fine, then he wants to switch the device over to SSID 2. He presses the button on the device and the ESP starts the configPortal. He selects another network and enters credentials, the ESP then writes "already connected, bailing out" and the ESP stays connected to SSID 1. Is there a way to switch networks when the device already is connected to one via the configPortal?

tablatronix, 2018-09-17:
Which branch?

henkep, 2018-09-17:
I am on main right now, I found the code that creates problem for me. In wifimanager.cpp on line 285:

`//fix for auto connect racing issue
  if (WiFi.status() == WL_CONNECTED) {
    DEBUG_WM(F("Already connected. Bailing out."));
    return WL_CONNECTED;
  }`