tzapu/WiFiManager #736: Custom Parameter with validation

millenseer, 2018-09-17:
Kindly asking for Feature Request

I was implementing WiFiManagerParameter for a project. Works really good! Until I discovered, that I accidentally left the MQTT-field empty. Which made it quite hard to change afterwards (had to switch of the Wifi-router to be able to connect to the ESP).

My suggestion is to integrate some basic validation method like:
- field is required, not empty. 
- field must contain IP address
- field must contain mail address
- field must be validated by custom callback
...

If validation fails, the first input form is re-displayed, with error message(s).
Of course very rudimentary.

Hence new parameter like:
WiFiManagerParameter custom_mqtt_server("server", "mqtt server", mqtt_server, 40, REQUIRED | IP_ADDRESS);

(Bit-operation not required, just an idea. Might be confusing)

**WiFimanager Branch/Release:**
- [x ] Master
- [ ] Development

**Esp8266/Esp32:**
- [x ] ESP8266
- [ ] ESP32

tablatronix, 2018-09-17:
just add required attributes to the html..

tablatronix, 2018-09-18:
- [ ] add param validation callbacks

millenseer, 2018-09-18:
The "required" attribute is not sufficient and requires manual changes in the WiFiManager.h, which is likely impractical when using the IDE and the library manager.
I guess the callback-method is the most practical way.

tablatronix, 2018-09-18:
well not really, the param class allows custom html attributes
`WiFiManagerParameter::WiFiManagerParameter(const char *id, const char *placeholder, const char *defaultValue, int length, const char *custom) {
`

millenseer, 2018-09-18:
Wow, development branch has greatly evolved!
Somehow I wasn’t aware about the custom html parameter. Now I read the class code, I understand what you’ve done by overloading. 

Thank you. I’ll be patiently awaiting new master release. Issue may be closed.

tablatronix, 2018-09-18:
for example
'  WiFiManagerParameter custom_ipaddress("input_ip", "input IP", "", 19,"pattern='\d{1,3}\.\d{1,3}\.\d{1,3}\.\d{1,3}'");
'

tablatronix, 2018-10-05:
you can now add custom param child classes #741

closed 2018-10-05 by tablatronix

AJTaylor79, 2022-10-12:
Hi @tablatronix just wondering how I would go about a custom validation and make sure that there in no spaces in text? Thanks