tzapu/WiFiManager #737: ADDING ELEMENTS TO THE HTML PAGE

diegoalbaga, 2018-09-19:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- x] Development

**Esp8266/Esp32:**
- [x] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [x] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [x] 2.4.0
- [ ] staging (master/dev)

### issue
Im trying to add a logo to de web server page but it would only let me modify the existing ones, how can I add new elements in the body and head (when I modify the html files it work on the test but not on the actual Sketch, in the strings file I can only modify the style of an existing component but I cant figure out how to add a new one) PLEASE HELP

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

diegoalbaga, 2018-09-20:
Thank I found were I needed to edit

closed 2018-09-20 by diegoalbaga

tablatronix, 2018-09-20:
You can also use customhtml params