tzapu/WiFiManager #739: Getting stuck in AP mode with unreliable WiFi connections

Duyvest, 2018-09-22:
Hi, 

I have a V2 nodemcu that does a Get request to a server to read values every 5 minutes.

I am using wifiManager 0.1.4.0 to establish the WiFi connection every 5 minutes.

The WiFi at devices location is a little bit unreliable, so when the wifi is down, the esp will create a AP/ captive portal and will no longer try to see the saved wifi network. 

Is there a work around for this situation?  I have tried wifiManager.setConfigPortalTimeout(180);  but this just restarts the captive portal instead of checking for the existing wifi network

`
    Serial.begin(115200);
    WiFiManager wifiManager;    
    wifiManager.setConfigPortalTimeout(30);  
    wifiManager.autoConnect("Du YVest");      
    Serial.begin(115200);
    WiFiManager wifiManager;    
    wifiManager.setConfigPortalTimeout(30);  
    wifiManager.autoConnect("Autoconnect");      
`
thanks

Mark

Van4ik, 2018-10-01:
I have a same question. How i can disable a AP, when i configured settings in station mode?

ubergeekseven, 2018-11-28:
Me too, having the same problem on ask the devices with WiFi manager.

tablatronix, 2018-11-28:
No idea what your code does, do it in your code. 
Or dont call autoconnect

Swiftnesses, 2018-12-09:
I also face this issue, I assume the timeout would recheck the saved network before starting :-(

salvq, 2019-09-20:
I would also very appreciate function that wifimanager in parallel keep checking for known AP credentails (in case router gets powerred off / restarted / firmware upgrade etc.) .

When no visible known network, you would end up having portal (AP mode) to enter new credential until the known network is visible, then board returns to client mode.

Is there any was to make this or workaround in sketch ?

Thanks

tablatronix, 2019-09-20:
esp already does this..

Swiftnesses, 2019-09-21:
> esp already does this..

The ESP does, yes, but with regards Wifi Manager, it doesn't work ideally when used with unstable connection (as per the issue).

tablatronix, 2019-09-21:
You can use ondemand non blocking configportal and start and stop it yourself..

Most of this can be done in user code now in development branch