tzapu/WiFiManager #740: Need to fill SSID second time if WebPortal

dontsovcmc, 2018-09-23:
### Basic Infos
#### Hardware
**WiFimanager Branch/Release:**
- [x] Development

**Esp8266/Esp32:**
- [X] ESP8266

**Hardware:			ESP-12e, esp01, esp25**
- [X] ESP01

1.8.0 version of https://github.com/platformio/platform-espressif8266

### Description

Run WebPortal and ESP connects to router successful.
ESP connects to router after deep sleep.
Run WebPortal (see SSID name in field) and didn't select SSID, password fields. Change only custom parameters. Press OK.

### Debug Messages

```
*WM: [2] -------------------- 
*WM: [3] Sent wifi save page 
*WM: [2] process connect 
*WM: [2] No ssid, skipping wifi 
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [3] -> captive.apple.com 
*WM: [2] <- Request redirected to captive portal 
*WM: [2] <- HTTP Root 
*WM: [3] -> 192.168.4.1 
*WM: [3] lastconxresult: WL_DISCONNECTED
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] Portal Timeout In 299 seconds
*WM: [2] WiFi Scan ASYNC completed in 2187 ms
*WM: [2] WiFi Scan ASYNC found: 8
*WM: [2] Portal Timeout In 269 seconds
*WM: [2] <- HTTP WiFi save  
*WM: [3] Method: POST
*WM: [2] Parameters 
*WM: [2] -------------------- 
```

### Source

I had a lot of additional code and little ESP code.

```
void loop() {
   
    if (setup) {   
        WiFiManager wm;
        wm.setConfigPortalTimeout(300);
        wm.setConnectTimeout(15);
        wm.startConfigPortal( AP_NAME );
    } 
    else {  
       WiFi.begin();  
       //do smth
    }
    ESP.deepSleep(0, RF_DEFAULT);
}
```
 
### Additional news:

1.  branch: hotfix, development
If I set this
``` 
        WiFi.persistent(false);  
        WiFi.disconnect(); 
        WiFi.mode(WIFI_AP);         
        WiFi.persistent(true);
```
before wm.startConfigPortal( AP_NAME ):
Run ESP startConfigPortal, refill saved SSID, password, push OK.
ESP *doesn't connect* to router, that has been already connected before. 

2. branch: development
If router turn off ESP connecting failed after timeout correctly. 

3. branch: development
Cyrillic labels shows nice in ConfigPortal.

### Idea about Integration tests:

I see a lot of Issues with abnormal behaviour in different modes, routers and some other stuff...
Maybe it possible to create Python/Bash script and ESP test sketch to preserve regression bugs in depending libraries and helps users to check their routers & code?

I think, we can turn on wi-fi active point in linux PC by command line commands, we can talk with ESP by Serial and check it states.

dontsovcmc, 2018-09-23:
If I don't write SSID when I open ConfigPortal, ESP doesn't connect. I think SSID&pwd are saved in memory and need not be filled every time when I open ConfigPortal...

tablatronix, 2019-01-31:
#782

dontsovcmc, 2019-11-27:
@tablatronix it's easy to fill down saved SSID & password to the input web form:

```
  pitem = FPSTR(HTTP_FORM_WIFI);
  pitem.replace(FPSTR(T_v), WiFi_SSID());  //move T_v from placeholder to value
  pitem.replace("{pwd}", WiFi_psk());  //add value='{pwd}' to template
```

```
placeholder='' value='{v}'>
type='password' placeholder='' value='{pwd}'
```
Now you can run Captive Portal for custom parameters changes without refilling SSID & pwd. 
I can make pull request if this way ok. I don't know why it wasn't done before.

tablatronix, 2019-11-27:
Is that not how it already works? What are you trying to solve?

dontsovcmc, 2019-11-28:
NO! 
If you fill nothing in ssid & pwd fields second run, ESP won't connect:
`*WM: No ssid, skipping wifi `

"Design is terrible". I don't know hidden ESP logic...
If SSID and password saved in ESP, we should show this fact to the user. Not in placeholder.

tablatronix, 2019-11-28:
What do you mean start configportal? 
If they are empty it skips saving, also sending the password back to the web is a security risk

tablatronix, 2019-11-28:
There is an issue for this I think already

tablatronix, 2019-11-28:
users are expected to know how esp works already and how it stores credentials.

dontsovcmc, 2019-11-28:
Hm.. about security I haven't think. Okey, we can send dots to interface value attribute. After Saved we call 
if (pwd.replace('.','').empty()) {
    pwd = WiFi_psk();
}

tablatronix, 2019-11-28:
I suggest stop using the wifi page to set parameters, and use the setup page now.

dontsovcmc, 2019-11-28:
I think it's overcode and avoiding the problem. We have a small UX task and you want to increase complexity of library...

tablatronix, 2019-11-30:
#974

dontsovcmc, 2019-11-30:
@tablatronix hm.. 
1. To show dots on the page is good.
2. I said, that previously saved password doesnt' work. In #974 you fill placeholder, not value. POST request parameter will be empty and ESP won't connect.

tablatronix, 2019-11-30:
Connect to what?

dontsovcmc, 2019-11-30:
To previously configurated Wi-Fi.

tablatronix, 2019-11-30:
When saving params?

dontsovcmc, 2019-11-30:
Ok, step by step.

1. First run startConfigPortal
2. Fill SSID, Password, click Save
3. Connect OK. Turn off device.

4. Second time run startConfigPortal
5. I see SSID in placeholder, nothing in Password. Click Save.
6. Connect FAILED. I see `*WM: No ssid, skipping wifi`

tablatronix, 2019-11-30:
Mine says its still connected, It never disconnects.

What settings are you using ?

dontsovcmc, 2019-11-30:
My device is autonomous, so I turn off power after startConfigPortal. 
So, If I want to call startConfigPortal second time (after reboot) I should fill password.

p.s. Device connect to wi-fi successfully after reboot, here is no problem.

tablatronix, 2019-11-30:
ok I see the problem

"Saving Credentials
Trying to connect ESP to network.
If it fails reconnect to AP to try again
"
If   // wm.setBreakAfterConfig(false); (default) and no wifi to save, the portal stays open.
The message above is wrong.

Ill make it a little better to understand.

If you want the portal to always close after save, use

  `wm.setBreakAfterConfig(true);`

dontsovcmc, 2019-11-30:
I know, that if connect failed (wrong password) it's also stays open - it's okey.

See this example:
https://github.com/tzapu/WiFiManager/blob/development/examples/ParamsChildClass/ParamsChildClass.ino  it's close to my sketch, but I turn off power in the loop.

So, If you run startConfigPortal second time, you should refill password.

tablatronix, 2019-11-30:
Yeah I need to fix this, setBreakAfterConfig used to do something different than what it said. So its confusing now.

dontsovcmc, 2019-11-30:
Great. For my opinion setShowPassword() is't needed.

tablatronix, 2019-11-30:
Using param saved page now, We will have to tweak that a bit, maybe auto redirect back to home, or add some more text or back and exit buttons etc

dontsovcmc, 2019-11-30:
i think:
if(_ssid == "" || String(_ssid) == WiFi_SSID())

tablatronix, 2019-11-30:
Closing this because its not really a bug, 
`Connect FAILED. I see *WM: No ssid, skipping wifi` means skipping wifi connect, has nothing to do with wifi failing. wifi reconnects when you close the configportal

closed 2019-11-30 by tablatronix