tzapu/WiFiManager #742: More than one set of retained WiFi credentials?

CaptClaude, 2018-09-23:
I have a question!
Having a good experience with WiFiManager on D1-Mini clones making portable @Cheerlights.
However: How many WiFi credential sets should be stored?
99% of the time the most recent are recalled and used but I seem to recall that I have seen more than one were retained.
Recent experiments done by moving between two different access points (out of range of each other) show that only the most recent is retained.

Am I missing something?  Can more than one set of credentials be retained (by adding more code?).

tablatronix, 2018-09-23:
Not with wifimanager, there are 2 prs that add multi ap support, and esp lib has multi support, but only 1 is retianed it is basically a list that it loops through then saves the one that works

CaptClaude, 2018-09-24:
Appreciate the quick response.
> 2 prs
Sorry for the slow uptake, but what are "prs"?
Having more than one is "nice to have", but certainly not a requirement.

tablatronix, 2018-09-24:
Pull requests

closed 2018-10-05 by tablatronix