tzapu/WiFiManager #744: cumston stored

miguerman, 2018-09-24:
char token chat id [?] = "-XXXXXXXX";
If I wanted to use customize and use to save the telegram token and the chat id that I must put in parentheses, for the example of the chat id of a telegram group that are 9 digits, [9].?
and for the telegram token?
Thank you

tablatronix, 2018-09-24:
Not sure what you are asking or saying

miguerman, 2018-09-24:
char mqtt_port [ 6 ] = " 8080 " ;
char blynk_token [ 34 ] = " YOUR_BLYNK_TOKEN " ;
orry, I want to use the example of addparameters of blynk, to use it with telegram and I see that where the port "8080" is reserved in parentheses [6] I understand that they are 6 digits but the port would be 5 digits, and for me to modify and save the telegram token and a chat id that are 9 digits, I think, that deneria ponet in parentheses.

miguerman, 2018-09-24:
char mqtt_port [6] = "8080"; 
char blynk_token [34] = "YOUR_BLYNK_TOKEN"; 
Lo siento, quiero usar el ejemplo de addparameters de blynk, para usarlo con telegram y veo que donde el puerto "8080" está reservado entre paréntesis [6] entiendo que son 6 dígitos pero el puerto sería de 5 dígitos, y para mí modificar y guardar el token de telegrama y una identificación de chat que son 9 dígitos, creo, eso debería ponerse entre paréntesis.

tablatronix, 2018-09-24:
The length on the string

closed 2018-10-05 by tablatronix