tzapu/WiFiManager #749: Create docs for next release, start wiki

tablatronix, 2018-10-05:
Looking for assistance

LucianaMarques, 2018-10-18:
Hi, I'd like to collaborate with this, how can I start?

tablatronix, 2018-10-18:
The code is commented, all the public functions should be.
The development issue contains an overview of all the new stuff #500

tablatronix, 2018-10-18:
I have started adding to readme, but it might need a major todo To organize it since the changes are major. It might be easier to start a wiki from scratch and reduce to readme the basics.

For example there is now nonblocking modes and on demand web portals, as welll as shared webserver pointer and parameter changes, aside from the entire refactor of strings to .h file

tablatronix, 2018-10-18:
If you want to start you can just start posting a comment here or fork and pr a .md doc and just start creating an outline of what you want to see as a user, its hard for me to do since I wrote it and know how it works?

tablatronix, 2022-02-13:
outline
https://github.com/tzapu/WiFiManager/wiki/Methods