tzapu/WiFiManager #750: save capture credentials how to

athleticmistud, 2018-10-10:
where in version 13 is the lines to save capture credentials so that if you turn on or off the board it will load last successful settings.  i am using ken taylors code but it has version 11 in it and is not saving the capture credentials so i have to keep entering it everytime

where or how do  i fix this

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [ ] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description

Problem description

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

tablatronix, 2018-10-10:
It is saved automatically when you save in configportal, if it is successful

athleticmistud, 2018-10-10:
no it does not save in version 11 which is the ken taylor version  and shows the ip its connected to.

so if someone   can help me where the lines are i need to add thanks

tablatronix, 2018-11-27:
The esp does the saving, wm does nothing but connect

matthewsellis, 2018-11-29:
> But in kens version it does not save so how to get that version. To save?

Having now googled this...

The esp arduino sdk automatically saves the ssid and psk to an area of the eeprom on a successful connection.  Check out the esp8266 wifi library docs for more details.

The old code in the wifimanager library to save ssid and password (psk) to eeprom has been commented out dpecifically because the esp wifi library already does this.

athleticmistud, 2018-11-29:
OK this is the issue ken’s version has the stuff marked out on it so you
can turn it back on and it will print the IP address which is what I need
but the problem is he based on the older generation so it has a lot of
errors with the Wi-Fi manager and does not work properly so I’m trying to
find out how to take the new Wi-Fi manager and add it so that it prints out
like ken’s version that IP address and if you go in an uncommon to the
things that are coming to the Audi does not make you still work so

tablatronix, 2018-11-29:
Just use the development branch

athleticmistud, 2018-12-19:
i dont know what that means im not a coder can someone help me so that i
can get it to work

hello  i use wifimanger and it works great and saves the cridentials so i
dont have to reenter them,  but i need to access it from an android as well
so i changed my wifimanger to ken taylors version.  that works great to
access with the androit and it prints out the ip address to access and
everything but IT DONT SAVE the cridentials and so i have to reenter
everytime i turn on my device.  i read in kens notes he did that for a
reason but does anyone have a fix for it so that i can get the best of both
worlds?  please help!