tzapu/WiFiManager #752: IMPROVEMENT: Allow callback or some other method when entering auto connect

Foozinator, 2018-10-12:
I'm working with an ESP8266 that has a small built-in OLED display.  I'd like to show the SSID and IP of the unit when it enters the AP-mode for connection input.  That way, I could just provide power to a "new" unit and not have to get a serial monitor to see what SSID + IP was used.

It looks like that might be simple enough to do if a callback was available for "about to enter AP-mode".

tablatronix, 2018-10-12:
#452

tablatronix, 2018-10-12:
```
    //called when AP mode and config portal is started
    void          setAPCallback( std::function<void(WiFiManager*)> func );
```

tablatronix, 2018-10-12:
It looks like that gets fired BEFORE starting the ap..

hmm

tablatronix, 2018-10-12:
I moved it after softap() in development branch

Should not break anything and allow what you want.

tablatronix, 2018-10-12:
1d0970a882b84d8840eb597785aae52e5c688605

tablatronix, 2018-10-12:
👊  Possible Breaking Change!

tablatronix, 2020-05-11:
Noone has complained... closing

closed 2020-05-11 by tablatronix