tzapu/WiFiManager #753: non blocking WM with custom parameters examples and documentation

scropion86, 2018-10-13:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

## POST SERIAL OUTPUT !

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [x] Development

**Esp8266/Esp32:**
- [x] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [x] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [x] 2.4.0
- [ ] staging (master/dev)

### Description

Add new example for non blocking WM with parameters
document the non blocking spicially saving custom parameters as i have tried for hours reading the closed issues and i still not able to figure it out

thanks for the great work 

### Settings in IDE

Module:  			custom build ESP-12F board

Additional libraries:  NON

tablatronix, 2018-10-13:
you parameters will need to be in the same scope as wm, which for non blocking is global for example

tablatronix, 2018-10-13:
2d021bbab82308ab7dd1e776e4a533578daf7956

tablatronix, 2018-10-13:
done

tablatronix, 2018-10-13:
if anyone has a better Idea how to create and destroy parameters and link to class let me know, I did not consider this when I made wm non blocking

tablatronix, 2018-10-13:
note documentation has not been started

closed 2018-10-13 by tablatronix