tzapu/WiFiManager #754: Not compatible with esp32?

moijerem59, 2018-10-14:
Arduino : 1.9.0-beta (Windows 10), Carte : "ESP32 Dev Module, Disabled, Minimal SPIFFS (Large APPS with OTA), QIO, 80MHz, 4MB (32Mb), 921600, None"

ATTENTION : la bibliothèque ESP32WebServer-master prétend être exécutable sur la (ou les) architecture(s) (ESP32) et peut être incompatible avec votre carte actuelle qui s'exécute sur (esp32).
In file included from C:\Users\Lenovo\Documents\Arduino\libraries\WIFIMANAGER-ESP32-master/WiFiManager.h:21:0,

                 from C:\Users\Lenovo\Documents\Arduino\libraries\WIFIMANAGER-ESP32-master\examples\AutoConnect\AutoConnect.ino:14:

C:\Users\Lenovo\Documents\Arduino\libraries\ESP32WebServer-master\src/ESP32WebServer.h:30:6: error: using typedef-name 'HTTPMethod' after 'enum'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

      ^

In file included from C:\Users\Lenovo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\WebServer\src/WebServer.h:30:0,

                 from C:\Users\Lenovo\Documents\Arduino\libraries\WIFIMANAGER-ESP32-master\examples\AutoConnect\AutoConnect.ino:12:

C:\Users\Lenovo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\WebServer\src/HTTP_Method.h:13:3: note: 'HTTPMethod' has a previous declaration here

 } HTTPMethod;

   ^

In file included from C:\Users\Lenovo\Documents\Arduino\libraries\WIFIMANAGER-ESP32-master/WiFiManager.h:21:0,

                 from C:\Users\Lenovo\Documents\Arduino\libraries\WIFIMANAGER-ESP32-master\examples\AutoConnect\AutoConnect.ino:14:

C:\Users\Lenovo\Documents\Arduino\libraries\ESP32WebServer-master\src/ESP32WebServer.h:30:19: error: redeclaration of 'HTTP_ANY'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

                   ^

In file included from C:\Users\Lenovo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\WebServer\src/WebServer.h:30:0,

                 from C:\Users\Lenovo\Documents\Arduino\libraries\WIFIMANAGER-ESP32-master\examples\AutoConnect\AutoConnect.ino:12:

C:\Users\Lenovo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\WebServer\src/HTTP_Method.h:12:3: note: previous declaration 'HTTPMethod HTTP_ANY'

   HTTP_ANY     = 0b01111111,

   ^

In file included from C:\Users\Lenovo\Documents\Arduino\libraries\WIFIMANAGER-ESP32-master/WiFiManager.h:21:0,

                 from C:\Users\Lenovo\Documents\Arduino\libraries\WIFIMANAGER-ESP32-master\examples\AutoConnect\AutoConnect.ino:14:

C:\Users\Lenovo\Documents\Arduino\libraries\ESP32WebServer-master\src/ESP32WebServer.h:30:29: error: redeclaration of 'HTTP_GET'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };

                             ^

In file included from C:\Users\Lenovo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\WebServer\src/WebServer.h:30:0,

                 from C:\Users\Lenovo\Documents\Arduino\libraries\WIFIMANAGER-ESP32-master\examples\AutoConnect\AutoConnect.ino:12:

C:\Users\Lenovo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\WebServer\src/HTTP_Method.h:5:3: note: previous declaration 'HTTPMethod HTTP_GET'

   HTTP_GET     = 0b00000001,

   ^

In file included from C:\Users\Lenovo\Documents\Arduino\libraries\WIFIMANAGER-ESP32-master/WiFiManager.h:21:0,

                 from C:\Users\Lenovo\Documents\Arduino\libraries\WIFIMANAGER-ESP32-master\examples\AutoConnect\AutoConnect.ino:14:

C:\Users\Lenovo\Documents\Arduino\libraries\ESP32WebServer-master\src/ESP32WebServer.h:30:39: error: redeclaration of 'HTTP_POST'

 enum HTTPMethod { HTTP_ANY, HTTP_GET, HTTP_POST, HTTP_PUT, HTTP_PATCH, HTTP_DELETE, HTTP_OPTIONS };
[... 149 lines cut ...]
C:\Users\Lenovo\Documents\Arduino\libraries\WIFIMANAGER-ESP32-master/WiFiManager.h:36:22: error: 'const char HTTP_HEAD []' redeclared as different kind of symbol

 const char HTTP_HEAD[] PROGMEM            = "<!DOCTYPE html><html lang=\"en\"><head><meta name=\"viewport\" content=\"width=device-width, initial-scale=1, user-scalable=no\"/><title>{v}</title>";

                      ^

In file included from C:\Users\Lenovo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\WebServer\src/WebServer.h:30:0,

                 from C:\Users\Lenovo\Documents\Arduino\libraries\WIFIMANAGER-ESP32-master\examples\AutoConnect\AutoConnect.ino:12:

C:\Users\Lenovo\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.0\libraries\WebServer\src/HTTP_Method.h:10:3: note: previous declaration 'HTTPMethod HTTP_HEAD'

   HTTP_HEAD    = 0b00100000,

   ^

exit status 1
Erreur de compilation pour la carte ESP32 Dev Module

Ce rapport pourrait être plus détaillé avec
l'option "Afficher les résultats détaillés de la compilation"
activée dans Fichier -> Préférences.

link webserver 
https://github.com/Pedroalbuquerque/ESP32WebServer

tablatronix, 2018-10-15:
You didnt fill out any information...

martinberlin, 2018-11-04:
I can confirm that development branch works perfectly on ESP-32 (Using now Heltec model Makerfocus 32)

closed 2018-11-04 by tablatronix