tzapu/WiFiManager #758: Wifi portal AP unreachable

nedimhx, 2018-10-25:
Hello, 

I have notice sometimes on some esp8266 devices, that I'm not unable to connect to Protal AP so I can configure wifi settings for esp to connect to router. But using that same esp on another location, I don't have problem.

When I'm unable to conncet to AP, after trying to connect, AP juste went away and again showed, secon case is, I have sucessfull connection to AP, but captive portal is not shown or if its shown, when I click scan networks it's just getin slow, and finnaly unable to connect to wifi network.

I have try completle to erase flas and upload code again from arduino ide but still the sam problem.

And after using that esp in other place like workshop its ok and captive portal working fine.

Can it be due to much wifi radio signals working on channel 1 and how is possible to set another radio signap channel in wifimanager.cpp 

Regards

tablatronix, 2018-10-25:
try development

nedimhx, 2018-10-25:
My latest version was be 0.11 and im updated to 0.14

Hope so this will fix issue.

nedimhx, 2018-10-25:
@tablatronix in the meantime how is possible to change Portal AP channal from 1 to some more rare radio signal channel ?

tablatronix, 2018-10-26:
What meantime, try the development branch to see if it has been fixed already.

You can also try the hotfixes branch which disabled sta mode to stabilize the softap

tablatronix, 2018-10-26:
You will have to manually change the code to start ap with channel, is your esp connected to a router?

fzoccara, 2018-10-30:
Hello everybody, I had the same problem, connecting to device wifi was not possibile, I'd experience this only in wifi crowd places.
Then I've fix my connection problem by changing the wifi channel (directly in the library).

In file Arduino\libraries\WiFiManager\WiFiManager.cpp by replace following rows:

```
  if (_apPassword != NULL) {
    WiFi.softAP(_apName, _apPassword);//password option
  } else {
    WiFi.softAP(_apName);
  }

``` 
with these rows: 
  
```
  if (_apPassword != NULL) {
    WiFi.softAP(_apName, _apPassword, 6);//password option
  } else {
    WiFi.softAP(_apName, NULL, 6);
  }
```

Is it possibile in a future version to make wifi channel configurable?

Best.
Francesco.

nedimhx, 2018-11-14:
@fzoccara After adding channel number as you told I got this error while compileing

Arduino: 1.8.4 (Windows 10), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, All Flash Contents, 115200"

C:\Users\nedim\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp: In member function 'void WiFiManager::setupConfigPortal()':

C:\Users\nedim\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp:134:27: error: invalid conversion from 'int' to 'const char*' [-fpermissive]

     WiFi.softAP(_apName, 3);

                           ^

In file included from C:\Users\nedim\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\libraries\ESP8266WiFi\src/ESP8266WiFi.h:35:0,

                 from C:\Users\nedim\Documents\Arduino\libraries\WiFiManager\WiFiManager.h:16,

                 from C:\Users\nedim\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp:13:

C:\Users\nedim\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.1\libraries\ESP8266WiFi\src/ESP8266WiFiAP.h:39:14: error:   initializing argument 2 of 'bool ESP8266WiFiAPClass::softAP(const char*, const char*, int, int, int)' [-fpermissive]

         bool softAP(const char* ssid, const char* passphrase = NULL, int channel = 1, int ssid_hidden = 0, int max_connection = 4);

              ^

exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

fzoccara, 2018-11-14:
Hi @nedimhx , you'd better check your row code:
` WiFi.softAP(_apName, 3);`

instead of 
` WiFi.softAP(_apName, NULL, 3);`

the error is eloquent:

> C:\Users\nedim\Documents\Arduino\libraries\WiFiManager\WiFiManager.cpp:134:27: error: invalid conversion from 'int' to 'const char*' [-fpermissive]

You are passing the channel id to the function softAP, where the function softAP expecting the password.

Best.
Francesco.

nedimhx, 2018-11-14:
@fzoccara thank you I forgot on that, thank you very much !

tablatronix, 2018-11-14:
I was testing a similar method in development branch, but I did not add any public methods to set channel, instead I added `_channelSync` which will set the ap to the same channel as sta to increase reliability.

I was thinking about another way to auto change channel if there was any way to detect failures, but I have not tested any of this, I still think the problem is not the channel, As I have not observed that much congestion with my wifi scanner, It is possible this is anecdotal and fixing some other problem

fzoccara, 2018-11-14:
Hi @tablatronix , 
At first I was thinking exactly like how you are thinking. However the same firmware on the same device was not working at the office only during crowd times, at home or in empty same office everything was regular.

Could be STA mode be itself part of the channel crowding? No further clue about any other cause, sorry. :)

Best.
Francesco.

tablatronix, 2018-11-14:
No but I seem to find it varies with devices as well, I have to be right next to my esp at work where I have 30+ aps, with my laptop, but my phone has no problem connecting. So I am not really sure what the deal is, I also noticed that the esp defaults to identifying itself as CN which sets invalid frequencies and could also cause problems with devices refusing to connect to it.

I added `setCountry` to development to try to help with that also with limited success.

So more testing for development branch and these 2 options would help us figure out a good way to fix these issues , and if crowding is indeed the problem, or if the esp is not rf calibrating properly on boot, or if the sdk has a bug or a limitation, I do know that soft+sta is problematic and it says so in the sdk

nedimhx, 2018-11-18:
@tablatronix This issue is definetly becouse of crowd places, I have test it again just for you, I use my bare esp 12e module wich purpose is to be RFID reader, but problem with AP was be all time, now I have reuploaded same code wich was be upload earlier, only modified wifi manager and set it on channel 3 instead of default 1, and module is working 100% fine, I have a couple of them wich I will test all.

By the way, can you guys sugest me wich is the best electric circute to solder for the best esp 12e stability and reliability ?

Paytah232, 2020-05-06:
When using wifiManager.autoConnect(ssid, pass), how can I try channel 3 instead of channel 1?

tablatronix, 2020-05-06:
all examples are in the DEV/ example

```
  // set channel
  // wm.setWiFiAPChannel(13);
```