tzapu/WiFiManager #762: setSaveParamsCallback() Hanging and not being able to save parameters

martinberlin, 2018-11-01:
#### Hardware
**WiFimanager Branch/Release:**
- [ ] Development

**Esp8266**
- [ ] ESP8266 Wemos D1

**ESP Core Version:	2.4.0, staging**
- [ ] staging (master/dev)

### Description

I have already a working version without using this new "param" in the menu on my FS2 Project and it's working fine. Also I though it could be a nice idea to let the users to edit the Setup without touching the connection but I'm struggling with it. 

The logic could be the same as setSaveConfigCallback() 

```
  wm.setBreakAfterConfig(true); // Without this saveConfigCallback does not get fired
  wm.setSaveConfigCallback(saveConfigCallback);
```
So this saveConfigCallback sets a flag like in your example shouldSaveConfig to true, and then IF (shouldSaveConfig) is true, then we save this in SPIFFS or whatever medium you use to save your configuration. 
Well, it turns out that for me in the current state acts the same as saveConfigCallback except that it hangs after returning. So I can never get to save the settings.

### Settings in IDE

Module:  			Wemos D1

Additional libraries: 

This is the feature branch where I tried to integrate this:
https://github.com/martinberlin/FS2/compare/feature/param?expand=1 

### Debug Messages

```
>>>>> Selected Camera model is: OV2640
SPIFFS file system mounted
Reading config file
{"timelapse":"30","slave_cam_ip":"","upload_host":"api.slosarek.eu","upload_path":"/camera-uploads/upload.php?f=2018","jpeg_size":"1600x1200"}
parsed json
>>>>>>>>>ONLINE Mode
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: timelapse
*WM: [2] Added Parameter: slave_cam_ip
*WM: [2] Added Parameter: upload_host
*WM: [2] Added Parameter: upload_path
*WM: [2] Added Parameter: jpeg_size
CAM can't get online. Entering config mode. Please connect to access point CAM-autoconnect
CAM-autoconnect
[CALLBACK] saveParamCallback fired -> should save config is TRUE
 ets Jan  8 2013,rst cause:2, boot mode:(1,6)
 ets Jan  8 2013,rst cause:4, boot mode:(1,6)
wdt reset
```

tablatronix, 2018-11-02:
You are not saving though, you are doing something odd here.

  if (memory.saveParamCallback) {
    wm.startConfigPortal(configModeAP);
  } else {
    wm.autoConnect(configModeAP);
  }

martinberlin, 2018-11-02:
Ok but then how it should be? 
I read should be only one of both. So I can only call autoConnect? But that one will connect automatically if we are on the same WiFi.
Can you please make me a 3 lines example.

But even if I’m doing this wrong it’s not supposed to hang or I’m thinking it wrong?

tablatronix, 2018-11-02:
not sure what you are wanting that to do, how are you saving your params now? just use that instead

if (memory.saveParamCallback) {
// get params and save them to whereever
}

no idea why you are trying to restart the configportal

tablatronix, 2018-11-02:
hmm I guess I am confused as to what this is doing

you are setting
`  shouldSaveConfig = true;` so that is good

what is purpose of
 `memory.saveParamCallback = false;`

tablatronix, 2018-11-02:
are you wanting to start the configportal if it was never setup before?
is that what that flag marks?

I would set that flag after you successfully save params.

tablatronix, 2018-11-02:
oh ok I think I might know what the problem is, save params does not stop the configportal.

I thought you were using non blocking mode, you are using blocking loop.

This makes sense now.

hmm, I may have to make some modifications to this to allow it to work with blocking mode, since there is no way to stop the configportal after the page gets served to user. Also you might not want to close the configportal.

let me think about this for a little bit, if you want to change your code to non blocking it will be a bit of work to move some stuff around.

tablatronix, 2018-11-02:
ok for now you will have to use exit button after save, it should work.

tablatronix, 2018-11-02:
I updated code, you can now call
 `wm.stopConfigPortal();` if you want to stop the configportal when blocking

so you can do

```C++
void saveParamCallback(){
  Serial.println("[CALLBACK] saveParamCallback fired");
  // wm.stopConfigPortal();
}
```

If you want to close the cp after saving params, until you can rewrite for non blocking

tablatronix, 2018-11-02:
do you know how to specify pins for OV2640 ?
I have one of these, I might be able to run your code on it.

[image]

tablatronix, 2018-11-02:
I see thanks.

There is not yet a way to pass the webserver into WM, but the wm webserver is a public shared pointer, so you can use it in your code, and add events to it etc.

see DEV/ondemandconfigportal example

```c++
void bindServerCallback(){
  wm.server->on("/custom",handleRoute);
  // wm.server->on("/info",handleRoute); // can override wm!
}
```

martinberlin, 2018-11-02:
Ah great I will try it out!  
After looking some pointers, I guess MOSI / MISO and Clock should be the same so maybe it's not so hard to get this working with a different camera. 
But of course it's time debugging and so on. I hope in 2 weeks will arrive my demo and you can connect this one I hope without to much effort first to some ESP8266 based board...And then if you want to ESP32 but I still didn't tested that. Today they arrived two Makerbot ESP32 boards so I will have some fun on the weekend :)

martinberlin, 2018-11-02:
Fantastic! Checked it two times, now it saves config correctly without hanging. Thanks a lot!
If anyone wants to see the implementation please check this branch on my camera project:
https://github.com/martinberlin/FS2/blob/feature/param/FasarekFS2/FasarekFS2.ino

closed 2018-11-02 by martinberlin