tzapu/WiFiManager #763: IMPROVEMENT: Predefined Credentials

SimonKlausLudwig, 2018-11-02:
I have the following scenario. If I am in my company I want that the ESP8266 connects to the company wifi. If I start the ESP8266 in a different environment I want that the WifiManager spins up an access point. Because we want to produce a lot of those devices it would be helpful to hardcode the predefined credentials in the code instead of using the Web interface.

If you are interested in such a feature i can help!

mrdc, 2020-01-07:
> 
> 
> wm uses default esp credentials, just use begin(ssid,pass); or some other esp method of doing this.

Unfortunately, it's not working: `'class WiFiManager' has no member named 'begin'`

tablatronix, 2020-01-07:
That esp class its not wm

mrdc, 2020-01-08:
@tablatronix 
> 
> 
> That esp class its not wm

Should I use WiFi.begin before WiFiManager code and WiFiManager will then understand that ESP is connected to WiFi and won't create AP?

mrdc, 2020-01-08:
Update:
```
WiFi.begin(SSID, PASS);
delay(500)
```
Above code works, WiFiManager is not making it's own AP to configure credentials, BUT WiFiManager fails to reconnect when predefined AP is off when ESP32 starts and when I restart ESP32 it fails to connect to predefined WiFi - WiFiManager makes the AP to configure WiFi credentials. To overcome the issues I need to go to WiFiManager portal -> Info -> Remove WiFi credentials -> then after restart it successfully connects to the predefined AP.

tablatronix, 2020-01-08:
erase NVS sounds corrupt credentials