tzapu/WiFiManager #764: Enhancement: Add input type param support

martinberlin, 2018-11-04:
I was thinking that for most of the people that are building their apps using this library as a base for WiFi Connectivity but also to Setup certain things like configuration using your "param" menu option would be a **nice addition** to have checkboxes and drop-down lists. 
You can, of course, think about it, since maybe it's out of scope. The main goal of this library is to provide a WiFi Manager that is already working very good. 

The first 2 elements that come to my mind while doing a configuration are:

1. ON/OFF checkbox
`<input type="checkbox" id="c1" name="c1" value="Turn on something"> Turn on something`

2. Options dropdown
`Select your car:
 <select>
  <option value="volvo">Volvo</option>
  <option value="saab">Saab</option>
  <option value="mercedes">Mercedes</option>
  <option value="audi">Audi</option>
</select>`
(Mine would be out of this price range ;)

tablatronix, 2018-11-04:
you should be able to add custom param support yourself without needing to modify the lib using the new param childclass capability

https://github.com/tzapu/WiFiManager/blob/development/examples/ParamsChildClass/ParamsChildClass.ino

martinberlin, 2018-11-06:
Thanks very nice example, will try it out and give you feedback about it. Happy that is already possible!

martinberlin, 2018-11-06:
I would like to see a input type="checkbox" example :) 
I still didn't find the time to try it. But I found sometime to make a SPIFFs file manager, you can try it out when your camera arrives!

tablatronix, 2018-11-06:
```C++
WiFiManagerParameter custom_field("customfieldid", "Custom Field Label", "Custom Field Value", customFieldLength,"placeholder=\"Custom Field Placeholder\" type=\"checkbox\"");
```
oh you can just add the type, but for select it will have to be totally different

tablatronix, 2018-11-06:
for custom html just add your own html

```c++
WiFiManagerParameter custom_html("<br/><label for='customfieldid'>Custom Field Label</label><input id='customfieldid' name='Custom Field Label' value='Custom Field Value'"); // custom html input
```

tablatronix, 2018-11-06:
if you want to make a child class to add additional types and add gettters to convert submission to whatever datatype you want, boolean , option id to text etc. then you have to use that example.

if you want to make a child class for option selects, you would need some kind of array and then push the options in then iterate and generate the custom html etc

tablatronix, 2018-11-06:
The only issue with this is that customhtml params do not have ids, so you cannot getValue etc and have to to it youself

tablatronix, 2018-11-06:
I updated advanced.ino example with a basic custom html input sample

martinberlin, 2018-11-07:
Great thank you I will try this out tonight and close this

martinberlin, 2018-11-19:
Works as expected. Good job!
Thanks a lot Shawn

closed 2018-11-19 by martinberlin