tzapu/WiFiManager #765: Compilation issue

au190, 2018-11-09:
Hi All

I would like to use some example in this code, but when Iam tring to compile I getting this error, how can I resolve this issue ? Is this because Iam using 2.4.2\libraries ??? Is not backward compatibile ? How can I use ?

Thx fror you answer.

Arduino: 1.8.5 (Windows 10), Board: "Generic ESP8266 Module, 80 MHz, Flash, ck, 26 MHz, 40MHz, QIO, 4M (1M SPIFFS), 2, v2 Lower Memory, Disabled, None, Sketch + WiFi Settings, 115200"

D:\uC\libraries\WiFiManager\WiFiManager.cpp: In member function 'void WiFiManager::setupConfigPortal()':

D:\uC\libraries\WiFiManager\WiFiManager.cpp:146:71: error: no matching function for call to 'ESP8266WebServer::on(String, std::_Bind_helper<false, void (WiFiManager::*)(), WiFiManager* const>::type)'

   server->on(String(F("/")), std::bind(&WiFiManager::handleRoot, this));

                                                                       ^

D:\uC\libraries\WiFiManager\WiFiManager.cpp:146:71: note: candidates are:

In file included from D:\uC\libraries\WiFiManager\WiFiManager.h:17:0,

                 from D:\uC\libraries\WiFiManager\WiFiManager.cpp:13:

D:\uC\libraries\ESP8266WebServer\src/ESP8266WebServer.h:79:8: note: void ESP8266WebServer::on(const char*, ESP8266WebServer::THandlerFunction)

   void on(const char* uri, THandlerFunction handler);

        ^

D:\uC\libraries\ESP8266WebServer\src/ESP8266WebServer.h:79:8: note:   no known conversion for argument 1 from 'String' to 'const char*'

D:\uC\libraries\ESP8266WebServer\src/ESP8266WebServer.h:80:8: note: void ESP8266WebServer::on(const char*, HTTPMethod, ESP8266WebServer::THandlerFunction)

   void on(const char* uri, HTTPMethod method, THandlerFunction fn);

        ^

D:\uC\libraries\ESP8266WebServer\src/ESP8266WebServer.h:80:8: note:   candidate expects 3 arguments, 2 provided

D:\uC\libraries\ESP8266WebServer\src/ESP8266WebServer.h:81:8: note: void ESP8266WebServer::on(const char*, HTTPMethod, ESP8266WebServer::THandlerFunction, ESP8266WebServer::THandlerFunction)

   void on(const char* uri, HTTPMethod method, THandlerFunction fn, THandlerFunction ufn);

        ^

D:\uC\libraries\ESP8266WebServer\src/ESP8266WebServer.h:81:8: note:   candidate expects 4 arguments, 2 provided

D:\uC\libraries\WiFiManager\WiFiManager.cpp:147:81: error: no matching function for call to 'ESP8266WebServer::on(String, std::_Bind_helper<false, void (WiFiManager::*)(unsigned char), WiFiManager* const, bool>::type)'

   server->on(String(F("/wifi")), std::bind(&WiFiManager::handleWifi, this, true));

                                                                                 ^

D:\uC\libraries\WiFiManager\WiFiManager.cpp:147:81: note: candidates are:

In file included from D:\uC\libraries\WiFiManager\WiFiManager.h:17:0,

                 from D:\uC\libraries\WiFiManager\WiFiManager.cpp:13:

D:\uC\libraries\ESP8266WebServer\src/ESP8266WebServer.h:79:8: note: void ESP8266WebServer::on(const char*, ESP8266WebServer::THandlerFunction)

   void on(const char* uri, THandlerFunction handler);

        ^

D:\uC\libraries\ESP8266WebServer\src/ESP8266WebServer.h:79:8: note:   no known conversion for argument 1 from 'String' to 'const char*'

D:\uC\libraries\ESP8266WebServer\src/ESP8266WebServer.h:80:8: note: void ESP8266WebServer::on(const char*, HTTPMethod, ESP8266WebServer::THandlerFunction)

   void on(const char* uri, HTTPMethod method, THandlerFunction fn);

[... 176 lines cut ...]
D:\uC\libraries\ESP8266WebServer\src/ESP8266WebServer.h:80:8: note: void ESP8266WebServer::on(const char*, HTTPMethod, ESP8266WebServer::THandlerFunction)

   void on(const char* uri, HTTPMethod method, THandlerFunction fn);

        ^

D:\uC\libraries\ESP8266WebServer\src/ESP8266WebServer.h:80:8: note:   candidate expects 3 arguments, 2 provided

D:\uC\libraries\ESP8266WebServer\src/ESP8266WebServer.h:81:8: note: void ESP8266WebServer::on(const char*, HTTPMethod, ESP8266WebServer::THandlerFunction, ESP8266WebServer::THandlerFunction)

   void on(const char* uri, HTTPMethod method, THandlerFunction fn, THandlerFunction ufn);

        ^

D:\uC\libraries\ESP8266WebServer\src/ESP8266WebServer.h:81:8: note:   candidate expects 4 arguments, 2 provided

Multiple libraries were found for "ESP8266WebServer.h"
 Used: D:\uC\libraries\ESP8266WebServer
 Not used: C:\Program Files\arduino-1.8.5\portable\packages\esp8266\hardware\esp8266\2.4.2\libraries\ESP8266WebServer
exit status 1
Error compiling for board Generic ESP8266 Module.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

tablatronix, 2018-11-09:
try hotfixes bramch

tablatronix, 2018-11-09:
#576

tablatronix, 2018-11-09:
if it works I will merge into master and release an update

mathieu-mp, 2018-12-17:
"hotfixes" branch solves the issue on Arduino IDE ! Thanks !

robertvorthman, 2019-04-01:
@tablatronix,  Hotfixes branch fixed this issue for me too.  Thanks.

closed 2020-05-11 by tablatronix

Michail1, 2020-08-21:
late 2020 and still not fixed other than with hotfix?
I get the issue with 2.00.00 and up, but
I don't get it with 0.15.0 and below.