tzapu/WiFiManager #767: Testing ota branch

martinberlin, 2018-11-17:
So far is a bit confusing. 

Scenario
Updated Wifi Manager to ota branch
/Arduino/libraries/WiFiManager$ git status
On branch ota
Your branch is up to date with 'origin/ota'

In my sketch : Added the "update" option to menu vector.

**What is expected to happen**
That update option appears when WiFi manager captive portal starts

**What is happening**
[image]
A new Erase option appears (But I didn't add erase to the menu array)

No update option appears. If I go to info, I see there is a /u Route, clicking on it I get:
File Not Found

URI: /u
Method: GET
Arguments: 0

Can you please explain in a short note how to test this ?
Thanks in advance

tablatronix, 2018-11-17:
what is your menu code?

martinberlin, 2018-11-17:
Is the same as you can see in the develop branch of the FS2 project. I copied it from your example if I remember well.

tablatronix, 2018-11-17:
I dont see any menu code in develop, did you push it?

martinberlin, 2018-11-18:
Good morning Shawn, 
Just added a feature/ota branch and added "update" in the menu:

https://github.com/martinberlin/FS2/commit/a8bf7ec6d7a8993c588d38e57efe0b51116fe514

tablatronix, 2018-11-18:
Fixed thanks!

martinberlin, 2018-11-19:
Works fantastic, thanks!!!

rodri16, 2019-11-27:
Hi, I am also testing OTA branch and when trying to upload a code I get an error, it says
```
*WM: [3] -> 192.168.4.1 
Update: Blink.ino
sleep disable
*WM: [3] -> 192.168.4.1 
.*WM: [3] -> 192.168.4.1 
ERROR[10]: Magic byte is wrong, not 0xE9
*WM: [2] <- Handle update don
```

tablatronix, 2019-11-27:
Ill take a look and merge in updates

tablatronix, 2019-11-27:
Did you check esp lib for this error?

rodri16, 2019-11-27:
I made it work sorry, it was a wrong file!!! I was trying tih .ino file and I have to use .bin
```
 *WM: [3] -> 192.168.4.1 
Update: Blink.ino.d1_mini.bin
sleep disable
*WM: [3] -> 192.168.4.1 
.*WM: [3] -> 192.168.4.1 
Updated: 261824 bytes
Rebooting...
*WM: [2] <- Handle update done 
*WM: [3] -> 192.168.4.1 
*WM: [1] [OTA] update ok 
del if0
station: 9e:f2:09:a1:3b:5a leave, AID = 1
rm 1
bcn 0
del if1
usl
```

Is OTA branch updated with last DEV changes?
Thanks @tablatronix

rodri16, 2019-12-12:
Hi @tablatronix 
I moved to dev branch but left "update" on the menu. Adding 

``` 
#define USEOTA
// enable OTA
#ifdef USEOTA
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
#endif
```

But update button doesn't apper in WM menu

tablatronix, 2019-12-12:
I gotta update that branch

tablatronix, 2019-12-12:
merged , let me test it real quick

tablatronix, 2019-12-12:
OTA is in the OTA branch, not dev

closed 2019-12-12 by tablatronix

tablatronix, 2019-12-12:
#981 
realized there was no issue for this