tzapu/WiFiManager #768: Create example, user config param for apname and password

tablatronix, 2018-11-19:
Create an example using custom params to display, save and read to flash? custom apname and password for wm ap mode.

tablatronix, 2018-11-19:
suggested options

- [ ] set cp apname
- [ ] set cp password or open

- [ ] setCaptivePortalEnable
- [ ] setConfigPortalTimeout
- [ ] setWebPortalClientCheck
- [ ] setEnableConfigPortal ( disable autoconnect cp, if you have a button etc )
- [ ] setCountry
- [ ] setHostname

serabiasin, 2019-07-27:
Hello, does this feature already implemented?  if not, i wanna try to implement the feature

tablatronix, 2019-07-27:
It is not

scropion86, 2019-07-28:
could you have a look to [this library](https://hieromon.github.io/AutoConnect/) it has a lot of similarities with WM and working on ESP32 also.

tablatronix, 2019-07-28:
Why? Wm already works on esp32..

scropion86, 2019-07-30:
i was not aware that Wm working with ESP32 as not mentioned in the Read me ,  after checking the development branch only has it mentioned.
nice to know