tzapu/WiFiManager #770: LED still on when router is down

zzxjl1, 2018-11-24:
Turn off the router after a sussessful connection, the Led is still on and the program seems unable to detect the current wifi status

tablatronix, 2018-11-24:
huh?

zzxjl1, 2018-11-24:
And it stucks in setup() when the router is off , how can i force it into loop() if i want to achieve some functions without the need of wifi？

zzxjl1, 2018-11-24:
I tried the timeout function , but it only reboots.

bigdolphin, 2018-11-24:
I think it is same with my issue and you should have to comment the line 2666 in the WiFiManager.cpp
https://github.com/tzapu/WiFiManager/issues/771

zzxjl1, 2018-11-25:
noop, i mean without any available ssid to connect it nerver tries to enter the loop()

tablatronix, 2018-11-25:
yeah that is how it is supposed to work if you call autoconnect(), it starts a configportal to configure using a blocking loop. development branch adds non blocking capability

zzxjl1, 2018-11-26:
Is there any ways to make it enter the loop while broadcasting the configportal

tablatronix, 2018-11-26:
See non blocking examples in development branch

closed 2018-11-26 by tablatronix