tzapu/WiFiManager #771: ESP32 reboot after Wifi is disconnected

bigdolphin, 2018-11-24:
After connected to a wifi, ESP32 reboots automatically if that wifi is disconnected.

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400e58e4  PS      : 0x00060630  A0      : 0x800e59a4  A1      : 0x3ffb3210  
A2      : 0xbf5caed8  A3      : 0x3f402a9a  A4      : 0x00060620  A5      : 0x3ffb1c70  
A6      : 0x00000000  A7      : 0x3ffb26a0  A8      : 0x800e58e4  A9      : 0x3ffb3200  
A10     : 0x00000005  A11     : 0x3f402a9f  A12     : 0x000000ff  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0xbf5caed8  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  

Backtrace: 0x400e58e4:0x3ffb3210 0x400e59a1:0x3ffb3230 0x400d9065:0x3ffb3250 0x400d9271:0x3ffb3290 0x400d53da:0x3ffb32b0 0x400e0415:0x3ffb3300 0x400e0596:0x3ffb33f0 0x400e0629:0x3ffb34f0 0x400e045f:0x3ffb3510 0x400e04bd:0x3ffb3600

Rebooting...
```

The reason at line 2666 in WiFiManager.cpp
DEBUG_WM(DEBUG_VERBOSE,"[EVENT]",event);

Comment that line solves the issue.

tablatronix, 2018-11-24:
I cannot reproduce

```
Current time: Sat Nov 24 20:52:39 2018
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:360] _eventCallback(): Reason: 200 - BEACON_TIMEOUT
*WM: [2] [EVENT] 5
*WM: [2] [EVENT] WIFI_REASON: 200
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:360] _eventCallback(): Reason: 201 - NO_AP_FOUND
*WM: [2] [EVENT] 5
*WM: [2] [EVENT] WIFI_REASON: 201
*WM: [2] [EVENT] WIFI_REASON: NO_AP_FOUND 
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:360] _eventCallback(): Reason: 201 - NO_AP_FOUND
*WM: [2] [EVENT] 5
*WM: [2] [EVENT] WIFI_REASON: 201
*WM: [2] [EVENT] WIFI_REASON: NO_AP_FOUND 
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:360] _eventCallback(): Reason: 201 - NO_AP_FOUND
*WM: [2] [EVENT] 5
*WM: [2] [EVENT] WIFI_REASON: 201
*WM: [2] [EVENT] WIFI_REASON: NO_AP_FOUND 
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:360] _eventCallback(): Reason: 201 - NO_AP_FOUND
*WM: [2] [EVENT] 5
*WM: [2] [EVENT] WIFI_REASON: 201
*WM: [2] [EVENT] WIFI_REASON: NO_AP_FOUND 
```

tablatronix, 2018-11-24:
when was the last time you pulled in development?

tablatronix, 2018-11-24:
ok got it now different sketch

```
WM: [3] unloading 
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:360] _eventCallback(): Reason: 200 - BEACON_TIMEOUT
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x00000000  PS      : 0x00060a30  A0      : 0x800dd348  A1      : 0x3ffb3370  
A2      : 0x3ffc4d44  A3      : 0x3f401bf3  A4      : 0x00060c20  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800dd295  A9      : 0x3ffb3360  
A10     : 0x3ffc4d44  A11     : 0x3f401bf3  A12     : 0x00000005  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000004  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffe  

Backtrace: 0x00000000:0x3ffb3370 0x400dd345:0x3ffb3390 0x400d72c9:0x3ffb33b0 0x400d74d5:0x3ffb33f0 0x400d376a:0x3ffb3410 0x400d2995:0x3ffb3460 0x400d2a29:0x3ffb3580

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7944
entry 0x40080310
*WM: [1] wifi 
*WM: [1] menuid found for  wifi
```

tablatronix, 2018-11-24:
event is called after wm is out of scope, must removeevent in deconstructor

`WiFi.removeEvent`

I have to add a way to pass the cb since it is a std::bind cb

tablatronix, 2018-11-25:
for now I am just checking a variable in scope, and bailing if its not true

CarnageMarkus, 2018-11-25:
Hi, I have the same problem -  its rebooting right after finishing `setup`. I pulled straight away and checked, but it still reboots right after `*WM: [3] unloading`

tablatronix, 2018-11-25:
after setup as in saving ?
I did not check that, let me test
I am assuming you pulled the fix in ?

tablatronix, 2019-02-27:
This broke esp32 async scannetworks, as the event never fires for some reason.

tablatronix, 2019-02-27:
I never set the darn variable to true!

commit ea1a0913 (2019-02-27): fix #771 again, I never set hasbegun to true!

should fix esp32 events not working, namely async scannetworks