tzapu/WiFiManager #777: WifiManager not working with fixed IP

eXadra, 2018-11-30:
Dear all.

Using this library with dhcp and parameters for the MQTT everithing works well (code based on your AutoConnectWithFSParametersAndCustomIP).
when I try to use with fixed ip, just adding the line 
 _wifiManager.setSTAStaticIPConfig(kIPAddress, kGateway, kSubnetMask);_
After reboot it returns to AP mode again, it seems that the wi parameters are not saved.
Check bellow 
 
_*WM: Configuring access point... 
*WM: ESP1795988
*WM: AP IP address: 
*WM: 192.168.4.1
*WM: HTTP server started
*WM: Handle root
*WM: Sent config page
*WM: WiFi save
*WM: Parameter
*WM: server
*WM: 192.168.1.246
*WM: Parameter
*WM: port
*WM: 1883
*WM: Parameter
*WM: User
*WM: exadra
*WM: Parameter
*WM: Password
*WM: pass
*WM: static ip
*WM: 192.168.1.12
*WM: static gateway
*WM: 192.168.1.254
*WM: static netmask
*WM: 255.255.255.0
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Custom STA IP/GW/Subnet
*WM: 192.168.1.12
*WM: Already connected. Bailing out.

0000047.795: WiFi connected. IP address: 192.168.1.12_

->> It says already connected and gets the IP 192.168.1.12 but thats not true, pinging it, this ip doesn´t exist. After reboot goes back to AP mode

Thanks

tablatronix, 2018-11-30:
you didnt fill out form

pvihang, 2019-07-31:
Dear @eXadra 
Did you find the solution to the Static IP issue using Wifimanager?
I am still not able to set Static IP using the AutoConnectwithStaticIp.ino  nor any luck with AutoConnectwithFSparametersandCustomIp.ino
On serial monitor it shows the IP which I have set, but I am unable to ping that IP. 
AutoConnect.ino for dhcp works great. But i need static IP configuration which is not working

Thanks

lizard79, 2019-10-12:
Did you find any solution? Have the same problem ESP-M3 , 8285.

Thanks

eyk107, 2019-12-02:
Could please someone post a working solution for fix dns and hostname?

tablatronix, 2019-12-03:
Use development branch

eyk107, 2019-12-03:
And Do you have a working example for IP, subnet, DNS and hostname?

tablatronix, 2019-12-04:
Not sure but the DEV/ example should have everything possible in it.