tzapu/WiFiManager #778: Accessing information of the connected wifi from the main sketch in the IDE

Bjack795, 2018-11-30:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

## POST SERIAL OUTPUT !

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [X] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [X] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [X] Other (Wemos D32 pro)

### Description
Hi! 
I'm using an ESP32 based board.
I need local_IP, gateway, subnet, dns, ssid and password of the connected wifi.
Could you please tell me how to access these information from the Arduino IDE or where to find them in the library files (to add some code to pass them to the Arduino IDE)?

tablatronix, 2018-11-30:
These are all standard functions for esp, has nothing to do with wm, note that on esp32 you cannot get these settings until wifi is connected.

Look in the esp docs

closed 2018-11-30 by tablatronix

Bjack795, 2018-11-30:
When the library scans and connects to a wifi collects those info to establish the connections isn't it?

tablatronix, 2018-11-30:
no the wifi just does not get initialized in esp32 like it does in esp8266 so the config is nor loaded until after wifi turns on, then you can read the network info

Bjack795, 2018-11-30:
Ok so I must refer to the Wifi.h functions to get those info

tablatronix, 2018-11-30:
They should all be in the docs already
but they are in wifigeneric and wifista wifiap

tablatronix, 2018-11-30:
you can look the wm source code for the info page if you want

Bjack795, 2018-11-30:
Ok I found all the data I needed thank you.
Offtopic: This library is saving the last wifi credential on the EEPROM memory?

tablatronix, 2018-12-01:
no esp does the saving