tzapu/WiFiManager #779: Library Intent

monkeytronics, 2018-12-02:
### Library Intent
Didn't know where to ask this. Apologies if not correct forum.

Like many who arrive at this repo, I'm attempting to connect my production ESP8266 devices to (unknown) wifi networks. Ideally, I want to run the WiFi commissioning within my own app, not have to redirect users to a third party app or a browser. Could I use this library to perform http like GET and POST operations on a server at a known address (192.168.4.1 is fine), in order to just run the actions in the background, without actually firing up a browser? Or am I grossly misunderstanding the overall intent of this library. In which case, I'd be eager for expert feedback as to what direction they would recommend... 

Or should I just directly use WiFi.softAP, with a password, limit number of connections to 1, and use SPIFFS to keep an up to date list of networks. Then run this when the user pushes a button. And very importantly, would this be secure? 

Thank you kindly for permitting my potentially naive and perhaps overly general question! But I feel that the research phase of what high level approach to take can save months of development time in the long run. 

#### Hardware
**WiFimanager Branch/Release:**
- [x ] Master
- [ ] Development

**Esp8266/Esp32:**
- [ x] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [x ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [x ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

tablatronix, 2018-12-02:
While wm does not yet have an api, you would be able to send post to it if you wanted, how do you find it your user still needs to connect to its ap

monkeytronics, 2018-12-02:
Yeah, totally. My thinking was to connect to the AP within the app, ideally by setting a hotspot ssid that is easy to pick identify. Or if necessary, instructing the user to find the hotspot themselves.  Then POST the credentials, followed by a complete command... 
body = {"ssid":"myHouse", "pw":"reallySecure", "id":"MyUniqueName"}; 
body = {"command":"complete"} 

Does this seem sensible? Or is there a better way?

tablatronix, 2018-12-02:
Afaik you can only connect to an ap via os not from an app on most systems

monkeytronics, 2018-12-02:
I was investigating a cordova hotspot plugin for use with Ionic Apps. 

https://github.com/hypery2k/cordova-hotspot-plugin/wiki/Ionic-usage

May or may not work. Worst case, ask the user to go find it.