tzapu/WiFiManager #781: REST API instead of HTML Pages

vip20, 2018-12-09:
I want to create my own UI so changing this to REST API instead of a HTML page would be of a help

ivanmonteiro, 2018-12-10:
Also interested

bkrajendra, 2019-04-20:
Any plans on this?
It will be very cool if all routes are available as rest api by default and there is way to disable front end UI. It will open lot of possibilities for UI. Also might reduce the memory footprint if UI is totally removed. I understand that basic intention of developer must have been making a standalone interface for wificonfig, but many things have changed now and there are ways we can better put UI on clientside with hybrid apps, native apps, PWA, or many other method.

i know all routes are available now also and we can definitely interface any front-end with it, but its not optimized the way it should be for API consumption.

@tzapu @vip20 @ivanmonteiro

vip20, 2019-04-20:
@bkrajendra  can you try this https://github.com/vip20/esp8266-wifimanager
Let me know if any issues

bkrajendra, 2019-04-20:
@vip20  i can try, but its always better to stick to original repo, as we can only get all updates in future from this repo. There are many available but its nice if API end points are available in original repo.!

I will look into the repo you provided and see if its helpful.

vip20, 2019-04-20:
@bkrajendra, I get it, but I couldn't wait longer as I had to carry on with my project

bkrajendra, 2019-04-20:
yes.. im also waiting from long time, this feature is must. I was thinking of doing directly by sending POST/GET request from ionic app and handling html parsing and app side. this way @tzapu  dont hve to change anything. but if little bit of HTML changed in WiFimanager it will break again. hence its better if simple JSON response is available from WM.

I dont if its possible or not in esp, but as im from web and mobile dev side, I know that on php server we can distinguish between ajax/rest client and browser through http headers. so its easy in server side to send selective response to client. 
I'll have to look into how it can be done in esp.

tablatronix, 2019-04-20:
Please post a schema or routes suggesiton, development branch lets you add your own endpoints as webserver is public so it might be easy to hack into user code for now.

bkrajendra, 2019-04-20:
Thanks @tablatronix for taking interest !
As i said i was looking for some ways to detect XHR and normal browser requests and I found this simple solution by detecting headers:

```
  if (server.hasHeader("X-Requested-With")) {
    Serial.print("Ajax Request Detected ");
    String xhr= server.header("X-Requested-With");
    Serial.println(xhr);
  // send JSON response 
  }else{
   // send HTML response
}
```
for this to work we need to have collectHeaders in setup before server.begin() :

```

  const char * headerkeys[] = {"User-Agent", "X-Requested-With"} ;
  size_t headerkeyssize = sizeof(headerkeys) / sizeof(char*);
  //ask server to track these headers
  server.collectHeaders(headerkeys, headerkeyssize);  
  server.begin();
```
Now we can send selective response to clients. If its browser then HTML response and if its a Android/ioS native client or a IONIC webview client then JSON response. 
Also http clients can send this ("User-Agent", "X-Requested-With")  header if they want to get JSON.

vip20, 2019-04-20:
@bkrajendra can you please let me know where to add this?
I am not able to connect to wifi using 0.14 please let me know if you face the same issue

bkrajendra, 2019-04-20:
@tablatronix 
for API routes it can be something like:
GET api/v1/wifi
GET api/v1/wifinoscan
POST api/v1/wifisave
GET api/v1/info
GET api/v1/restart
GET api/v1/erase
GET api/v1/status

**or better keep the routes same as now and use my previous solution to detect XHR/Browser. This will save lot of effort and avoid creating separate handlers. We can keep same handlers and just implement request detection.**

bkrajendra, 2019-04-20:
@vip20 i was suggesting header solution to have REST client to get JSON response, so that we can connect through app. There are some other issues like keeping portal open etc. thats totally different issues. Also im not getting what exactly you are asking? I generally use Development branch only, even if it breaks sometime :)

vip20, 2019-04-20:
@bkrajendra I'll try to implement it tomorrow, I badly need this feature, have you ever tried with firebase connection with esp8266?

bkrajendra, 2019-04-20:
No @vip20 , I generally use my own VPS servers with cloud and mqtt on it.

tablatronix, 2019-04-20:
I would rather implement an api endpoint and write custom api class, than have to add ajax checking to each callback just to keep it cleaner and be able to add features to api easier

rvhani, 2020-01-01:
I'm waiting for this. Rest API. sory for 1-liner.

vip20, 2020-01-03:
@bkrajendra have any luck on this?

laurensV, 2022-11-27:
2.5 years later but would still be great to have this feature!

laurensV, 2022-11-27:
Or is there a different example of a wifimanger that exposes API endpoints so it can be setup through a mohile app for example?

tablatronix, 2022-11-28:
You can make your own api in user code