tzapu/WiFiManager #782: Hotfixes

dontsovcmc, 2018-12-11:
### Basic Infos
If run Web Portal second time (after good connect first time) it doesn't connect.
This happend both with refill Wi-Fi & pass and no fill previous Wi-Fi.
If I fill another Wi-Fi - ESP connects successfully.

#### Hardware
**WiFimanager Branch/Release:**
- [x ] HotFixes
**Esp8266/Esp32:**
- [x] ESP8266

**Hardware:**
- [x ] ESP01

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)
i don't know. 
``` .platformio/framework-arduioespressif8266/platform.txt:
# ESP8266 platform
# ------------------------------
name=ESP8266 Modules
version=2.4.2
```

### Debug Messages

```
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Status:
*WM: 6
*WM: [ERROR] WiFi.begin res:
*WM: 0
*WM: Waiting for connection result with time out
*WM: Connection timed out
*WM: Connection result:
*WM: 0
*WM: Failed to connect.
*WM: Request redirected to captive portal
```
If I write another wi-fi title - OK

```
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Status:
*WM: 0
*WM: [ERROR] WiFi.begin res:
*WM: 6
*WM: Waiting for connection result with time out
*WM: Connection result:
*WM: 3
000:00:02:05:224  NOTICE    (AP) : Connected to wifi. Save settings, go to sleep
```

If I don't fill Wi-Fi & Pass:
```
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Status:
*WM: 0
*WM: Using last saved values, should be faster
*WM: Waiting for connection result with time out
*WM: Connection timed out
*WM: Connection result:
*WM: 0
*WM: Failed to connect.

// hard turned off ESP (EN = LOW)

// Wake up ESP to transmitting data - It connects successfully! 
rl␀l��|␀�l�|␂␌␌␌�␌l�␌b|��␂�␒�r�bl␌b��nn�lnn���␌b␜p��lrlrlp�n�␐␂␌␌�␌l␌��␌␌␌b␌n�|␂l�␌␌�b��nn�␀l��l`␂�␒␒nn␌l`␂␎␂nr���n␌␌b␌�␎l␎r��n␌␌b␌�␎l�bbb␌␌���l`␂��n�␂000:00:00:00:050  NOTICE    (ESP) : Booted
000:00:00:00:119  NOTICE    (WIF) : Starting Wifi
000:00:00:00:124  NOTICE    (WIF) : Wifi status: 6
....
000:00:00:03:332  NOTICE    (WIF) : Wifi status: 6
000:00:00:03:532  NOTICE    (WIF) : Connected, got IP address: 10.1.30.45
000:00:00:04:251  NOTICE    (BLK) : disconnected
```

tablatronix, 2018-12-12:
Try hotfixes

tablatronix, 2018-12-12:
Ohh lol this is that bug in hotfixes from the other issue we were discussing

tablatronix, 2018-12-12:
Ill try to fix , I have been extremly busy

tablatronix, 2019-01-31:
What do you mean if you run webportal a second time ? ONdemand ?

Egor2101, 2019-01-31:
after a reset.

tablatronix, 2019-01-31:
you mean when you savewifi or just after reset it wont connect ?

tablatronix, 2019-01-31:
I might have to fix this, it does nothing unless you set `wifiManager.setBreakAfterConfig(true);`

tablatronix, 2019-02-07:
Anyone test this.

dontsovcmc, 2019-02-07:
I try to test, thank you very much!
So, I need to add `wifiManager.setBreakAfterConfig(true);` to my code before start portal?

tablatronix, 2019-02-07:
yes if you want to use the callback, for now

dontsovcmc, 2019-02-11:
I think, it doesn't work =(. Reconnect only if you enter wrong password, got fail, write correct.

```
connect to wi-fi OK
....
turn on setup portal for the second time.
write same ssid & password

*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Status:
*WM: 0
*WM: [ERROR] WiFi.begin res:
*WM: 0
*WM: Waiting for connection result with time out
*WM: Connection timed out
*WM: Connection result: 
*WM: 0
*WM: Failed to connect.
*WM: Request redirected to captive portal
....
*WM: Request redirected to captive portal
*WM: WiFi save
....

write incorrect password

....
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Status:
*WM: 0
*WM: [ERROR] WiFi.begin res:
*WM: 6
*WM: Waiting for connection result with time out
*WM: Connection result: 
*WM: 4
*WM: Failed to connect.
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: WiFi save
....

write correct password

....
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Status:
*WM: 4
*WM: [ERROR] WiFi.begin res:
*WM: 6
*WM: Waiting for connection result with time out
*WM: Connection result: 
*WM: 3

OK

```

tablatronix, 2019-02-12:
I thought you were entering no wifi?

dontsovcmc, 2019-02-12:
update log: enter =(.
If I don't enter wi-fi I see `No ssid, skipping wifi`

tablatronix, 2019-02-12:
Yeah its supposed to

tablatronix, 2019-02-12:
Wait so if you enter the wifi credentials they do not work? That is another open issue already

dontsovcmc, 2019-02-12:
Yes, second time filled wi-fi doesnt connect.
you need to fill wrong password, wait error in log, change to correct password to connect.

Yes,if fields are empty no wifi connection.

tablatronix, 2019-02-12:
ok well you should be able to just leave it empty now at least.

There is a problem reconnecting once mode changes, and the only fix was to add a disconnect, I will check the other issue and see what the problem is, I thought there was a workaround in hotfixes for that already.

dontsovcmc, 2019-02-12:
I imagine to build ESP test board:
3 NodeMCU:
1. Active Point, controled by script
2. WIFI client - http requests, controled by script
3. Target - with testing firmware. Flashing, reseting by script

We will be able to run scripts for different tasks

dontsovcmc, 2019-03-31:
Try to test connect 2nd time: Failed.
ESP-12
Branch: hotfix

```
*WM: Sent config page
*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Status:
*WM: 0
*WM: [ERROR] WiFi.begin res:
*WM: 6
*WM: Waiting for connection result with time out
scandone
switch to channel 3
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with don, channel 3
dhcp client start...
ip:192.168.1.50,mask:255.255.255.0,gw:192.168.1.1
*WM: Connection result: 
*WM: 3
station: 28:cf:e9:1e:ea:91 leave, AID = 1
rm 1
bcn 0
del if1
pm open,type:2 0
mode : sta(60:01:94:29:66:e8)
*WM: freeing allocated params!

The second time run Config Portal:

*WM: Sent config page
*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Status:
*WM: 0
*WM: [ERROR] WiFi.begin res:
*WM: 0
*WM: Waiting for connection result with time out
*WM: Connection timed out
*WM: Connection result: 
*WM: 0
*WM: Failed to connect.
```

dontsovcmc, 2019-04-01:
@tablatronix look, I think, I found a bug in ESP8266WiFiSTA!

Wow.. I debugging ESP8266WiFiSTA.cpp, so this is not your bugs.. 

I try to understand: is configurations are same?  I print structures as hex and see, that 

```
    Serial.println("conf_compare:");
    char *p = (char*)(void*)&conf_compare;
    char b[5] = { 0, 0, 0, 0, 0};
    for (unsigned int i = 0; i < sizeof(struct station_config); i++) {
        snprintf(b, 3, "%02X", *p);
        p++;
        Serial.print(b);
    }

    Serial.println();
    Serial.println("conf:");
    p = (char*)(void*)&conf;
    for (unsigned int i = 0; i < sizeof(struct station_config); i++) {
        snprintf(b, 3, "%02X", *p);
        p++;
        Serial.print(b);
    }
```
1. conf_compare & conf have a lot of garbage cause doesn't `setmem 0` when creating.
2. the function sta_config_equal doesn't works good, because there are no some fields of station_config!!

Look! Is the same SSID & password Wi-Fi structures are same?
```
conf_compare:
646F6E0000000000000000000000000000000000000000000000000000000000
3232327177654153445A58430000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

000107BEF5C074400000000000000000001000000

conf:
646F6E0000000000000000000000000000000000000000000000000000000000
3232327177654153445A58430000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000

00000000000000000810000000200000001000000
```

```
***************************************************************************************************************************************
Obsolete PIO Core v3.6.0 is used (previous was 3.6.6)
Please remove multiple PIO Cores from a system:
http://docs.platformio.org/page/faq.html#multiple-pio-cores-in-a-system
***************************************************************************************************************************************
--- Miniterm on /dev/cu.SLAB_USBtoUART  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
s$␀l��|␀�l�|␃␄␄␌�␌d�␜p�{�␃�␒�r�#�␌#�p␡�n�loN���␌c␜p��$r$rdp�o�␐␃␌␄�␄l␄��␄␌␄b␌N�|␂l�␄␌�b��oN�␃d��␀$`␃�␒␒oN␌d`␃␏␂nr���N␌␄c␄�␎$␎{��o␄␌b␌�␏l�␒l�p␂���␒�`␂��n�␂
SDK:3.0.0-dev(c0f7b44)/Core:2.5.0=20500000/lwIP:STABLE-2_1_2_RELEASE/glue:1.1/BearSSL:6778687
Booted
scandone
6
scandone
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 
6

connected with don, channel 3
dhcp client start...
6
6
ip:192.168.1.50,mask:255.255.255.0,gw:192.168.1.1
[... 260 lines cut ...]
*WM: Waiting for connection result with time out
scandone
switch to channel 3
state: 0 -> 2 (b0)
state: 2 -> 3 (0)
state: 3 -> 5 (10)
add 0
aid 2
cnt 

connected with don, channel 3
dhcp client start...
ip:192.168.1.50,mask:255.255.255.0,gw:192.168.1.1
*WM: Connection result: 
*WM: 3
station: 28:cf:e9:1e:ea:91 leave, AID = 1
rm 1
bcn 0
del if1
pm open,type:2 0
mode : sta(60:01:94:29:66:e8)
*WM: freeing allocated params!

--- exit ---
```

dontsovcmc, 2019-04-01:
I return sta_config_equal back: FAILED, so the problem with comparing structures!

```
*WM: WiFi save
*WM: Sent wifi save page
*WM: Connecting to new AP
*WM: Connecting as wifi client...
*WM: Status:
*WM: 6
mode : sta(60:01:94:29:66:e8) + softAP(62:01:94:29:66:e8)
add if0
conf_compare:
646F6E00000000000000000000000000000000000000000000000000000000003232327177654153445A58430000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000107BEF5C074400000000000000000001000000
conf:
646F6E00000000000000000000000000000000000000000000000000000000003232327177654153445A5843000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000810000000200000001000000*WM: [ERROR] WiFi.begin res:
*WM: 0
*WM: wifi_get_opmode(): 
*WM: 3
*WM: Persistent:
*WM: 1
*WM: Waiting for connection result with time out
*WM: Connection timed out
*WM: Connection result: 
*WM: 0
*WM: Failed to connect.
```

dontsovcmc, 2019-04-01:
there are a lot of discussions about this function here: 
https://github.com/tzapu/WiFiManager/issues/702
https://github.com/esp8266/Arduino/issues/3641

dontsovcmc, 2019-04-01:
accepted! =)

tablatronix, 2019-04-01:
nice catch, thank you

closed 2019-08-30 by tablatronix