tzapu/WiFiManager #784: not uploading to nodeMCU

sabishaw, 2018-12-12:
Hi everybody!
when i m tryna upload this code 
`#include <ESP8266WiFi.h>          //ESP8266 Core WiFi Library (you most likely already have this in your sketch)
#include <DNSServer.h>            //Local DNS Server used for redirecting all requests to the configuration portal
#include <ESP8266WebServer.h>     //Local WebServer used to serve the configuration portal
#include <WiFiManager.h>   

void setup() {
 
WiFiManager wifiManager;

//first parameter is name of access point, second is the password

wifiManager.autoConnect("corph","corph@786");

//wifiManager.autoConnect();

}

void loop() {

  
}

 `
it shows this error message
`Arduino: 1.8.7 (Windows Store 1.8.15.0) (Windows 10), Board: "NodeMCU 0.9 (ESP-12 Module), 80 MHz, Flash, 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

libraries\WiFiManager-master\WiFiManager.cpp.o: In function `String::operator+=(String const&)':

C:\Users\MEHARR\Documents\Arduino\libraries\WiFiManager-master/WiFiManager.cpp:891: undefined reference to `WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool'

libraries\WiFiManager-master\WiFiManager.cpp.o: In function `WiFiManager::startWPS()':

C:\Users\MEHARR\Documents\Arduino\libraries\WiFiManager-master/WiFiManager.cpp:891: undefined reference to `WPS_is_unavailable_in_this_configuration__Please_check_FAQ_or_board_generator_tool'

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board NodeMCU 0.9 (ESP-12 Module).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
`
when i try to  upload any other program..it goes fine.....could you please suggest reason and solution for this... thanks in advance

tablatronix, 2018-12-12:
Old esp library?

sabishaw, 2018-12-12:
> Old esp library?
could you please tell me what do you mean by that

teddibaer, 2018-12-14:
I faced the same issue (using platformio with esp8266).

It seems that platformio is using v2.4.2 of the espressif8266 library. From the readme:

About WPS
From release 2.4.2 and ahead, not using WPS will give an exra ~4.5KB in heap.
In release 2.4.2 only, WPS is disabled by default and the board generator is required to enable it:
./tools/boards.txt.py --allowWPS --boardsgen

**For platformIO (and maybe other build environments), you will also need to add the build flag: -D NO_EXTRA_4K_HEAP**

This manual selection is not needed starting from 2.5.0 (and in git version). WPS is always available, and not using it will give an extra ~4.5KB compared to releases until 2.4.1 included.

tablatronix, 2018-12-14:
I think I added checks for this in development already

tablatronix, 2018-12-14:
#646 
Hotfixes applied

closed 2018-12-14 by tablatronix

sabishaw, 2018-12-14:
Thanks for replying
teddibaer   i actually solved this problem in my case....and quite honestly bro,i didnt understand your post.....whatever
in my  case if you see error message carefully it speaks about  line 891  in wifimanager.cpp
but when you open header file of wifimanager it has only 804 lines....
now it means it was reffering to some other file with same name...
then i searched all the files with name wifimanager and deleted them
and installed new one from github and guess what?
it worked!!!! yeeey...
i hope it might help ya if you have exactly same problem...
Auf Weidezien