tzapu/WiFiManager #786: Intermitent error only on PlatformIO (connects, portal -> loop)

martinberlin, 2018-12-16:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [x] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [x] ESP32

**ESP Core Version:	2.4.0, staging**
- [x ] staging (master/dev)

### Description

I'm struggling to find out why compiling WifiManager in platformIO gives me this strange bug. It's simply that one time it connects, next time does not, and so on.
I thought that it was the connectTimeout that was too short but of course, it's not that. 
NOTE: This happens only compiling stuff with platformIO (Does not matter what ESP32 board)
In other projects, I compiled this with Arduino and I didn't have this problem at all. So what I would like to know without making you lose your precious time is how can I help to debug this to the bone and find out what is causing this.

NOTE: I'm trying this with the camera project but I completely stripped it out, leaving only:
wm.autoConnect(configModeAP);

Not any other thing that may interfere just left it like this to test what may be happening.
Question: wm.setConnectTimeout(10); // what are there millis ? 

`*WM: [1] AutoConnect
*WM: [2] ESP32 event handler enabled
*WM: [2] Connecting as wifi client...
*WM: [2] setSTAConfig static ip not set
*WM: [3] WIFI station disconnect
*WM: [1] Connecting to saved AP: KabelBox-A210
*WM: [3] WiFi station enable
*WM: [2] 10000 ms connectTimeout set
*WM: [2] 10000 ms timeout, waiting for connect...
*WM: [2] .
*WM: [2] .
*WM: [2] .
*WM: [2] .
*WM: [2] .
*WM: [2] .
*WM: [2] .
*WM: [2] .
*WM: [2] .
*WM: [2] .
*WM: [2] .
*WM: [2] .
*WM: [2] .
*WM: [2] .
*WM: [2] .
*WM: [2] .
*WM: [2] .
*WM: [2] .
*WM: [2] .
*WM: [2] .
*WM: [2] Connection result: WL_CONNECT_FAILED
*WM: [3] lastconxresulttmp: WL_IDLE_STATUS
*WM: [3] lastconxresult: WL_CONNECT_FAILED
*WM: [1] AutoConnect: FAILED
*WM: [3] WIFI station disconnect
*WM: [3] WiFi station enable
*WM: [2] Disabling STA
*WM: [2] Enabling AP
*WM: [1] StartAP with SSID:  CAM-autoconnect
*WM: [2] AP has anonymous access!
*WM: [1] AP IP address: 192.168.4.1`

tablatronix, 2018-12-16:
I would assume you are using a different esp32 verison, Check info page and compare the 2, I cannot think of anythink else that would cause difference with pio, unless you have debug on or some other change

tablatronix, 2018-12-16:
Same router?

martinberlin, 2018-12-17:
Will do it. It does not matter if it's at home or with my mobile hotspot. Same behaviour. One time connects, next time opens Portal, and so on. 
So I will just capture a snapshot of the info page on PlataformIO and then compile the same with Arduino and do the same. There has to be something fishy, maybe low Heap memory ? 
The only difference is that on PlataformIO I configured the FS different and I'm leaving about 5MB for SPIFFs. 
https://github.com/martinberlin/FS32/blob/master/fs_partition.csv

tablatronix, 2018-12-17:
Very strange, what if you do not call autoconnect and just begin() and connected loop

tablatronix, 2018-12-17:
I just remembered someone else posted a similar issue and it also made no sense. Ill see if I can find it, maybe look around or search for similar words if you have time. It was every other restart it would not connect, or only if unplugged it would connect. Some weird issue like that.

martinberlin, 2018-12-18:
YES! Got it. 
It's like this: On a quick restart, I get always this intermitent YES / NO (Open portal) situation.

If I wait at least 15 seconds, leaving it off, and then starting it again it always connect. Between 12/14 seconds it's the least I can wait. I can live with it is not a problem! 
But I would like to know why. And specially why it does not happen when I compile this with Arduino !

tablatronix, 2018-12-18:
Did you ever check to make sure you are using the same verison, maybe you have arduino set to use a different lwip version ? or have debugging on or off, debugging can reveal or hide bugs sometimes

martinberlin, 2018-12-18:
Didn't found time to test more but I just checked, WiFi Manager is same version exactly (development)
but ESP32 core is not.
https://github.com/platformio/platform-espressif32.git#feature/stage -> PlataformIO
Arduino -> Default esp32 v. 1.0.0 
That can be the thing! Anyways I will debug it more. You can close this one if it bothers to have many things open. It's actually not a bug in your library but an issue compiling it in different platforms (If I'm thinking right)

martinberlin, 2018-12-19:
Yes, feature/stage but for Espressif32 core. Thanks for the info!

tablatronix, 2018-12-19:
oops yeah thats what I meant.
I will try to reproduce, and reread what I think the problem was in that other issue. What happens if you  reboot the eso not reset, same thing ?

tablatronix, 2018-12-20:
have you tried with basic example ?

tablatronix, 2018-12-20:
can you save the 2 bin files and put them somewhere?

tablatronix, 2018-12-20:
- [ ] test basic example
- [ ] test another router ( or hotpsot )

tablatronix, 2018-12-20:
Cannot reproduce, I can hit reset button and connect everytime.

martinberlin, 2018-12-20:
>have you tried with basic example ?
No, but I will try with a new ESP32 that Heltec people sent me, also without the camera project. I'm suspecting that some of my stuff may be interferencing with WM.

>Is your wm object global?
Yes, it's actually in a clon of the Camera project but using the esp32 version

>can you save the 2 bin files and put them somewhere?
Sure, I still don't know how to do this with plataformIO but I will google it ;)

tablatronix, 2018-12-20:
yeah it will be saved in your build folder

martinberlin, 2019-01-06:
Hello, long time no talk, I'm already recovered and ready to start a new year of fun.

https://storage.luckycloud.de/f/704030ca95374d399663/

Uploaded whole .pioenvs/lolin_d32 folder. Using a Wemos ESP32 since I do not need any fancy display since I'm using my own TFT. I really like this version with double SPI. It works really fast and uploads a 150 Kb picture in about 4 seconds with good WiFi.
By the way is still like I described before, just waiting about 20/30 seconds between disconnects it works fine, leaving this intermittent error only when you want to restart fast (Is annoying anyways but I can live with it)
ONE important thing since you mentioned: platform = https://github.com/platformio/platform-espressif8266.git#feature/stage

I'm using ESP32 in this thing for now :)

tablatronix, 2019-01-06:
Thanks I will have a look

martinberlin, 2019-01-09:
An ESP32-WROOM-32D  is coming my way so I will try to debug this with PlataformIO, keeping you updated

martinberlin, 2019-01-11:
I figured out something interesting with this one. 
I was using for this test a [Heltec ESP32 Wifi](https://docs.platformio.org/en/latest/boards/espressif32/heltec_wifi_kit_32.html) but compiling because I got some SPI erros it using board: 
[env:lolin_d32]
platform = espressif32
board = lolin_d32

Now I changed this to the right board:
`[env:heltec_wifi_kit_32]
platform = https://github.com/platformio/platform-espressif32.git
board = heltec_wifi_kit_32
framework = arduino
monitor_speed = 115200
; set frequency to 80/ 160MHz
board_build.f_cpu = 160000000L`

And the problem seems to have disappeared. So it was a configuration error on my side, I guess different boards have slight different settings for the WiFi, can it be ?
Anyways not an error from WiFi Manager itself so closing this one.

closed 2019-01-11 by martinberlin

tablatronix, 2019-01-11:
I would not think so, unless the memory is wrong. you can check boards file for difference, or verbose output and check the build command

tablatronix, 2019-01-11:
I use the esp32 dev or the doit board definition, most designs follow these pinouts

1technophile, 2020-04-08:
I had also this issue with ESP32, here is more info about it:
https://github.com/espressif/arduino-esp32/issues/2501

tablatronix, 2020-04-08:
This issue is so sporadic its hard to determine if its an actual bug, I still am going with a race condition, I experienced it once and it immediately resolved itself.

1technophile, 2020-04-08:
In my side I was able to reproduce it one time out of 2, in all the case it is not a WifiManager issue, it is more related to ESP32 Arduino Framework or the ESP32