tzapu/WiFiManager #788: Can't open "Configure WiFi" page with my phone

AlcorDust, 2018-12-18:
Hello, i'm using wifiManager on my NodeMCU. Here's the details:

Board: nodeMCU 1.0
4M (3M SPIFFS)
Upload speed 115300
Frequency 80 MHz

I uploaded the example "AutoConnect", and i'm able to connect and see the main page on my smartphone. The problem is that i can't open the page "Configure wifi" when i press the button. I tried to connect to the AP with my laptop and It works correctly. Is it a smartphone problem? I have a Huawei P9 Lite.

tablatronix, 2018-12-18:
sounds like maybe there is an html problem, try development branch ?

AlcorDust, 2018-12-19:
Hi, i tried development branch. Using the "Basic" example sketch i can't even open the main page. I'm going to try older versions of the library!

AlcorDust, 2018-12-19:
Here's what i did:
- I tried with older versions but nothing changed.
- I used 2 ESP8266, with the same result.
- If i open the serial monitor i read this:
`*WM: 192.168.4.1
*WM: HTTP server started
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Request redirected to captive portal
*WM: Handle root
*WM: Handle root
*WM: Handle root
*WM: Scan done
*WM: DUP AP: WOW FI - FASTWEB
*WM: DUP AP: Vodafone-Guest
*WM: FASTWEB-1-55D705
*WM: -58
*WM: WOW FI - FASTWEB
*WM: -58
*WM: InfostradaWiFi-012943
*WM: -61
*WM: TIM-21030679
*WM: -72
*WM: Vodafone-A45210187
*WM: -72
*WM: Vodafone-WiFi
*WM: -73
*WM: Vodafone-Guest
*WM: -75
*WM: FASTWEB-B836B7
*WM: -87
*WM: Vodafone-33898704
*WM: -88
*WM: Vodafone-A74575006
*WM: -92
*WM: DIRECT-22-HP OfficeJet 4650
*WM: -92
*WM: Sent config page
*WM: Request redirected to captive portal
*WM: Handle root
`

tablatronix, 2018-12-19:
Yeah sounds like a problem with captive portal, can you connect to ap fine and goto the ip address?

Could be a security setting, some android has locked down captive portals for some ips

tablatronix, 2018-12-19:
Develoment has the most debugging, you should test without captiveportal first