tzapu/WiFiManager #789: Adding new/additional WiFiParam to existing App causes crash at configuration

gduprey, 2018-12-19:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [X] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [X] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [] Other
- [X] ESP32 OLED WEMOS/WROVER dev board

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)
- [X] 1.0.0 (ESP32)

### Description

I have an existing app with no parameters (no WiFiManagerParameters).  Configuration works great (using either autoConnect() or startConfigPortal()).  I then add a single new parameter to the application, flash it and when I try to connect to the the config portal, the ESP crashes (below).  

If I add a resetSettings() call, then things work fine (I can connect, configure and see the parameter in the portal).  But if I store settings and then reboot the chip and re-enter config mode (startConfigPortal()), when I attempt to connect to the ESP AP, it crashes again.

If I remove the parameter, everything works as expected (I can configure and reconfigure many times with no crashes).

The crash happens after the config portal appears as an AP, at the moment you attempt to connect to it (and by connect, I mean join the AP's SSID from my phones WiFi settings page).  So you never get a chance to even attempt to connect to the actual config portal web page with a browser - the WiFi connection alone initiates the crash.

I'm running on an ESP32/WROVER using the development branch WiFiManager.

### Settings in IDE

Module:  			ESP32 WROVER 

Additional libraries: 

### Sketch

```cpp

#include <WiFiManager.h>

WiFiManagerParameter * custom_led_count;

void setup() {
  char ledCountBuffer[16];
  WiFiManager wm;

  // Configure WiFiManager
  wm.setDebugOutput(false);
  //wm.resetSettings();
  wm.setConfigPortalTimeout(300);

  // Set configurables
  sprintf(ledCountBuffer, "%d", stripLEDCount);
  custom_led_count = new WiFiManagerParameter("ledCount", "Strip LED Count", ledCountBuffer, 16);
  wm.addParameter(custom_led_count);

  // Set AP settings
  wm.setAPStaticIPConfig(IPAddress(192, 168, 254, 1), IPAddress(192, 168, 254, 1), IPAddress(255, 255, 255, 0));

  // See if we should unconditionally enter config mode vs. auto-mode
  pinMode(CONFIG_BUTTON, INPUT_PULLUP);
  if (digitalRead(CONFIG_BUTTON) == LOW) {
    wifiStarted = wm.startConfigPortal(nodeName);
  } else {
    // Attempt connection or start auto-config
    wifiStarted = wm.autoConnect(nodeName);
  }
}

void loop() {

}
```

### Debug Messages

```
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
Booting
*WM: [3] allocating params bytes: 20
*WM: [2] Added Parameter: ledCount
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  NET_LED
*WM: [1] Custom AP IP/GW/Subnet: 
*WM: [2] AP has anonymous access! 
*WM: [1] AP IP address: 192.168.254.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.254.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] Config Portal Running, blocking, waiting for clients... 
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x00000000  PS      : 0x00060030  A0      : 0x801050b9  A1      : 0x3ffb6270  
A2      : 0x3ffbcb40  A3      : 0x3ffbc328  A4      : 0x3ffbbd8c  A5      : 0x3ffbcb20  
A6      : 0x02fea8c0  A7      : 0x0cfea8c0  A8      : 0x80104f58  A9      : 0x3ffb6230  
A10     : 0x3ffbcb50  A11     : 0x3ffbc328  A12     : 0x3ffb627c  A13     : 0x00000044  
A14     : 0x00000001  A15     : 0x00000006  SAR     : 0x00000010  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000  

Backtrace: 0x00000000:0x3ffb6270 0x401050b6:0x3ffb62b0 0x4010d06d:0x3ffb62d0 0x4010fc19:0x3ffb6310 0x4011635e:0x3ffb6330 0x40104479:0x3ffb6350

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
Booting
```

pieman64, 2018-12-19:
@gduprey did you look at an example with parameters as your sketch looks to be missing quite a bit of code?

pieman64, 2018-12-19:
Parameters appear 2 or 3 times in the example code. Try one of them.

gduprey, 2018-12-19:
Yep - the block I pasted was stripped down, but was not the final one (missed a define and two variables).  I have that and will post/edit the above when I get back home later today.  

There may also be some interaction with the Arduino preferences API - I need to play with a few configs there too as something seems a little off (but not reliably, of course...)

gduprey, 2018-12-19:
Well, I'm embarrassed.  In the process of trimming things down and trying to make an oh-so-perfect report, I cut out (partially) use of preferences - specifically, calling .end() on them. That made something in the stack get sick (I assume after the .end(), attempts to access prefs would fail from in WiFi/WiFiManager).

Lesson learned - don't close out prefs before WiFIManager has its time with them.

Sorry for the noise.

closed 2018-12-19 by gduprey

tablatronix, 2018-12-19:
what is preferences API ?

tablatronix, 2018-12-19:
interesting, this must be new, I will check it out, pretty cool