tzapu/WiFiManager #790: Wifimanager does not work with esp8266webserver

alemori, 2018-12-20:
## PLEASE TRY DEVELOPMENT BRANCH before submitting bugs on release or master, in case they were already fixed. ##

## POST SERIAL OUTPUT !

Issues without basic info will be ignored or closed!

Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.rst

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.rst

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [ ] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description

Problem description

### Settings in IDE

Module:  			NodeMcu, Wemos D1

Additional libraries: 

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

tablatronix, 2018-12-20:
probably because wm uses it and you cannot use it twice

alemori, 2018-12-20:
and how can I use a server.send that works?

tablatronix, 2018-12-20:
the development branch allows it

alemori, 2018-12-20:
How is it done?  Can you give me an example?  Thank you

tablatronix, 2018-12-21:
#461

tablatronix, 2018-12-21:
Development branch dev/ondemandconfigportal should have an example in it

alemori, 2018-12-21:
I went to Development branch dev/ondemandconfigportal, but only wifimanager is used there. I also wanted to compile the .ino and gave me errors, I guess the error is mine, because I do not know how to work with the development code. What I need is to be able to write the html interface using the ESP8266webserver facility. If there is any example that you can give me, I would appreciate it. I copy the code that I'm trying to make work, and for example when I invoke 192.168.4.1/pepepe, it doesn't give me anything back.

#include <FS.h>          // this needs to be first, or it all crashes and burns...
#include <WiFiManager.h> // https://github.com/tzapu/WiFiManager
#include <ESP8266WebServer.h>
#include <ESP8266mDNS.h>
#include "index.h"
ESP8266WebServer server2(81);
String s1 = MAIN_page; //Read HTML contents

void handleRoot() {
  //server.setContentLength(CONTENT_LENGTH_UNKNOWN);
  server.send(200, "text/plain", "hello from esp8266!");
}
void pepe(){
 Serial.print("Entro a handleRoot");
 //server.setContentLength(CONTENT_LENGTH_UNKNOWN);
 server.send(200, "text/html", s1); //Send web page
}
void handleReset() {
  //server.setContentLength(CONTENT_LENGTH_UNKNOWN);
  server.send(200, "text/plain", "Resetting!");
    //Serial.println("diconnecting client and wifi");
    //client.disconnect();
    wifi_station_disconnect();

  WiFiManager wifiManager;
  wifiManager.resetSettings();
  ESP.reset();
  
}

void handleNotFound(){
  String message = "File Not Found\n\n";
  message += "URI: ";
  message += server2.uri();
  message += "\nMethod: ";
  message += (server2.method() == HTTP_GET)?"GET":"POST";
  message += "\nArguments: ";
  message += server2.args();
  message += "\n";
  for (uint8_t i=0; i<server2.args(); i++){
    message += " " + server2.argName(i) + ": " + server2.arg(i) + "\n";
  }
  server2.send(404, "text/plain", message);
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println();
 // Serial.setDebugOutput(true);
  

  //WiFiManager
  //Local intialization. Once its business is done, there is no need to keep it around
  WiFiManager wifiManager;

  //exit after config instead of connecting
  wifiManager.setBreakAfterConfig(true);
  //reset settings - for testing
  //wifiManager.resetSettings();

   
  //tries to connect to last known settings
  //if it does not connect it starts an access point with the specified name
  //here  "AutoConnectAP" with password "password"
  //and goes into a blocking loop awaiting configuration
  if (!wifiManager.autoConnect("AutoConnectAP", "password")) {
    Serial.println("failed to connect, we should reset as see if it connects");
    delay(3000);
    ESP.restart();
    delay(5000);
  }

  //if you get here you have connected to the WiFi
  Serial.println("");
  Serial.print("Connected to ");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  if (MDNS.begin("esp8266")) {
    Serial.println("MDNS responder started");
  }

  server.on("/", handleRoot);

  server.on("/inline", [](){
    server.send(200, "text/plain", "this works as well");
  });
  server.on("/resetwifi", handleReset);
  server.on("/pepe", pepe);
  server.onNotFound(handleNotFound);

//  server.begin();
  server.begin();
  Serial.println("HTTP server started");

}

void loop() {
  // put your main code here, to run repeatedly:
  server.handleClient();

}

=====================
INDEX.h
======================

 const char MAIN_page[] PROGMEM = R"=====(
<! DOCTYPE html> 
<html> 
<head>
<meta name="viewport" content="width=device-width, initial-scale=1.0">
<title> LED Control </title> 
<style>
body {
    background-color: lightblue;
}
h1 {
    color: black;
    text-align: center;
    font-family: verdana;
    font-size: 190%;
}
h3 {
    color: black;
    text-align: center;
    font: verdana;
    font-size: 190%;
    
}
p {
    font-family: verdana;
    font-size: 120%;
    text-align: center;
}
p1 {
    font-family: verdana;
    font-size: 120%;
    text-align: center;
}
.button {
    background-color: #4CAF50; /* Green */
    border: none;
    color: white;
    padding: 15px 32px;
    text-align: center;
    text-decoration: none;
    display: inline-block;
    font-size: 120%;
}
.button2 {
    background-color: rgb(0, 128, 255); /* Green */
    border: none;
    color: white;
    padding: 15px 32px;
    text-align: center;
    text-decoration: none;
    display: inline-block;
    font-size: 120%;
}
#rcorners1 {
    border-radius: 25px;
    background: #73AD21;
    padding: 20px; 
    width: 200px;
    height: 150px; 
}
</style>
</head>
<div>
<body>
<form action='/inicio' method='post'>
  <h3> DATOS DEL ATLETA</h3>
  <br><br/>
  <p1> Nombre/Equipo <input name="Nombre" type="text" maxlength="50" size="50" /> </p1>
  <br><br/>
  <p1><input type="radio" name="genero" id = "radio1" value="femenino" checked> Femenino<br></p1>
  <br><br/>
  <p1><input type="radio" name="genero" id = "radio2" value="masculino"> Masculino<br></p1>
  <br><br/>
  <p1> Edad <input name="edad" type="text" maxlength="2" size="2" /> </p1>
  <br><br/><br><br/>
  <p><input name="bt" class = 'button' type="submit" /><p>
  </form> 
</div>
<div>
<script>
  var xhttp = new XMLHttpRequest();
   
  xhttp.open("GET", "fechahora?valorfecha="+new Date().toLocaleString(), true);
  xhttp.send();
</script>
</div>

</body>
</html>
)=====";

alemori, 2018-12-24:
I'm going to wait for it to be incorporated into the library, since I don't want to use the development banch. Thank you

sarmadsohaib, 2022-08-03:
I am late on this but experiencing exactly the same problem. Can you please tell me how to change the webserver from private to public in the code?