tzapu/WiFiManager #791: Quality Assurance Testing Needed

tablatronix, 2018-12-20:
If anyone wants to help out with development release, Please QA it, and target any issues labelled QA thanks.

This will help test things that I cannot , or have not yet, and let me close issues and mark releases faster

dontsovcmc, 2019-03-31:
Please, add info about debugging WiFiManager, ESPCore and so on.

dontsovcmc, 2019-08-30:
I can't imagine how huge errors and tickets are opened for WifiManager....
I think the most important think - to make release. But it's impossible with a lot of changes. 
There is hard to understand bug versioning with unnamed 'development' branch...

Maybe we can separate issues by:
- critical bugs, need help bugs
- refactors
- soft features
- big features

Choose issues for next release and fix them.
What do you think about it?

tablatronix, 2019-08-30:
development is practically a complete refactor, it will be a major release, v1 or 2

tablatronix, 2019-08-30:
an overview of all changes is documented, and all code is commented.

https://github.com/tzapu/WiFiManager/issues/500