tzapu/WiFiManager #792: Different behaviour Wifimanager on 2 different ESP8266 12E

randoniaina, 2018-12-20:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [x ] Master
- [ ] Development

**Esp8266/Esp32:**
- [x ] ESP8266
- [ ] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [x ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ x] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description

Hello everyone,

I do face a very weird situation. I have 2 ESP8266 12e. 
- One that I worked on early 2018, on which I have Wifimanager library (don't remember the version since I upgrade this version since then).
 = > this board is connected to a wifi router

- Two other boards on which I've bee working for 2 months and on which there are Wifimanager library also (version 0.14.0)
= > These 2 boards are also connected to the SAME WIFI ROUTER.

The problem is that the last 2 boards launch the config portal randomly (for an unknown reason) : the Wifi routeur is always on WHEREAS the former board (on which I worked on early 2018) does NOT launch the config portal randomly.

The expected behaviour would be the 1st behaviour (that on the former board).

Can someone please help ? I cannot figure it out.

Here the code of the older board 

```
#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
#include <WiFiClient.h>

//needed for library
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h> 

/*
Measuring AC Current Using ACS712
*/
const double sensorIn = A0;
int LED_PIN = 4;
int CONFIG_PIN = 12;
int statut_relay_actuel;
int statut_relay_avant;
int statut_line;

int mVperAmp = 30; // use 100 for 20A Module and 66 for 30A Module

double Voltage = 0;
double VRMS = 0;
double AmpsRMS = 0;

//const char *ssid = "DLINKAP";
//const char *password = "Disney";

const char* host     = "mydomain.com"; // Your domain
[... 466 lines cut ...]
    //WiFiManager
    //Local intialization. Once its business is done, there is no need to keep it around
    WiFiManager wifiManager;
    //reset saved settings
    wifiManager.resetSettings();
    
    //set custom ip for portal
    wifiManager.setAPStaticIPConfig(IPAddress(10,0,1,1), IPAddress(10,0,1,1), IPAddress(255,255,255,0));
 
    //fetches ssid and pass from eeprom and tries to connect
    //if it does not connect it starts an access point with the specified name
    //here  "AutoConnectAP"
    //and goes into a blocking loop awaiting configuration
    wifiManager.autoConnect("AutoConnectAPEmetteur");
    //or use this for auto generated name ESP + ChipID
    //wifiManager.autoConnect();
 
    //if you get here you have connected to the WiFi
    Serial.println("connected...yeey :)");

}

//####################### FIN ###################"

```

tablatronix, 2018-12-20:
have you erased the flash ?

randoniaina, 2018-12-20:
Yes I did every time I upload a sketch I tick "erase all flash" on the ide config. Then I enter the portalconfig and select the wifi router. 

-------- Message d'origine --------De : Shawn A <notifications@github.com> Date : 20/12/2018  22:06  (GMT+03:00) À : tzapu/WiFiManager <WiFiManager@noreply.github.com> Cc : randoniaina <andoniainarajaona@gmail.com>, Author <author@noreply.github.com> Objet : Re: [tzapu/WiFiManager] Different behaviour Wifimanager on 2 different ESP8266 12E (#792) 
have you erased the flash ?

—
[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/tzapu/WiFiManager/issues/792#issuecomment-449103287",
"url": "https://github.com/tzapu/WiFiManager/issues/792#issuecomment-449103287",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

randoniaina, 2018-12-20:
What should I do if I want the following behaviour :- let the esp try to Connect to the wifi routeur even if it fails (do not let the esp open the portalconfig) - open the the portalconfig only and only if I push a button
Regards 

-------- Message d'origine --------De : Shawn A <notifications@github.com> Date : 20/12/2018  22:06  (GMT+03:00) À : tzapu/WiFiManager <WiFiManager@noreply.github.com> Cc : randoniaina <andoniainarajaona@gmail.com>, Author <author@noreply.github.com> Objet : Re: [tzapu/WiFiManager] Different behaviour Wifimanager on 2 different ESP8266 12E (#792) 
have you erased the flash ?

—
[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/tzapu/WiFiManager/issues/792#issuecomment-449103287",
"url": "https://github.com/tzapu/WiFiManager/issues/792#issuecomment-449103287",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]

tablatronix, 2018-12-20:
dont use autoconnect()

tablatronix, 2018-12-20:
Is it possible you have wore out the flash ram by writing too many times?

randoniaina, 2018-12-21:
No cause this behaviour started happening despite the board was New. 
Can you confirm one thing ? Autoconnect is about connecting to a wifi routeur, Nothing to do with either or not there is internet provided by the wifi routeur, is that correct ? 
Cause sometimes even if the routeur is on, there is internet connection problem. 
Regards 

-------- Message d'origine --------De : Shawn A <notifications@github.com> Date : 20/12/2018  22:48  (GMT+03:00) À : tzapu/WiFiManager <WiFiManager@noreply.github.com> Cc : randoniaina <andoniainarajaona@gmail.com>, Author <author@noreply.github.com> Objet : Re: [tzapu/WiFiManager] Different behaviour Wifimanager on 2 different ESP8266 12E (#792) 
Is it possible you have wore out the flash ram by writing too many times?

tablatronix, 2018-12-21:
yeah sounds like something with the router I think there is some kind of problem disconnecting and reconnecting perhaps, the only thing I can think of is increase connecttimeout, or add your own loop ir delay and see if it fixes by letting the router reset the lost connections.

Have you checked to make sure your esp is not wdt reseting, causing these resets

Have you tried disabling sleep
`WiFi.setSleepMode(WIFI_NONE_SLEEP); // disable wifi sleep`

randoniaina, 2018-12-22:
Seems like the router sometimes has some pblms assigning ip addresses. But the weird part is why the old board is not portalconfig showing whereas the newer ones are ? 

Envoyé depuis mon appareil mobile Samsung.
null

randoniaina, 2018-12-22:
What is wifi sleep mode ? How to check the wdt reset or not ? 
Tx in advance 

Envoyé depuis mon appareil mobile Samsung.
null