tzapu/WiFiManager #793: ESP8266_RTOS_SDK

misha-marinenko-official, 2018-12-26:
Can some one port this to https://github.com/espressif/ESP8266_RTOS_SDK ?

Torniyo, 2020-05-02:
Hi there.... I'm kinda super new in all of this and I've been trying something that is, probably, not possible. tablatonix you may help me with this one....
I've installed ESP8266_RTOS_ SDK and cloned several versions of the "Wifimanager" in order to try make it work..... most of the developers uses Arduino IDE... and they have different problems or achieve success.... I'm not even sure of in which part of failure do I dwell... ja!
Is it possible to the wifimanager not being prepared for the ESP8266_RTOS_SDK and all my effort has been futile?  Please... If you're kind enough, let me know with your most rude and sincere words.... maybe I'm trying to fly too close to the sun... and... I don't even know how to fly yet!!! Thanks in advanced... a complicated and confused newbie from Argentina...

tablatronix, 2020-05-03:
I have no experience with rtos