tzapu/WiFiManager #801: Example - my approach to including wifimanager into the code

TinajaLabs, 2019-01-08:
Showing this here for feedback.  I love wifimanager but I find it difficult to intersperse the various pieces of wifiManager code into my own code. 

This is an example that I use for a tip-cup rain gauge.  The wifiManager part allows a user to set the local wifi station as well as the mqtt server (node-red) and part of the mqtt topic for location info.

The main parts of wifimanager have been broken into the following methods which make it easier for me to set up for other sensors.  I'm including features like `On Demand button`,  `reset with long press down`, `SPIFFS`. 
`I would like to know if I'm covering all the bases properly or if there's a better way to do this`.  Not an expert with the Arduino C-like code:
* saveConfigCallback
* runWifiManager
* loadConfigFile
* saveConfigFile

`The thought here is that perhaps there could be some examples that show these pieces as methods rather than large chunks of code in the setup or loop methods.`

Also, as I mentioned, I also collect some location info in the wifiManager page which is to be used to create the appropriate mqtt topics when using the sensors in different locations.  Typical topics might look like this:
* farm/barn/fenceline/WemosD1Mini/rain/state/value
* farm/barn/fenceline/WemosD1Mini/rssi/state/value

The first part (farm/barn/fenceline) is collected by a wifiManager parameter and then constructed into the final topic string in a separate method, setRainTopics. `I would like to know if this is a reasonable approach or maybe a better way.`

Thanks for any tips,
Chris.

```/*
   TinajaLabs.com
  ---------------------------------------------------------------
   wfmt-rainguage

   Attributes:
   - WifiManager
   - FS
   - MQTT

   - RSSI: dbm
   - davis rain guage: tip bucket rain guage
     outputs triggers indicating the bucket has tipped out 0.01 inches of rain

   Arduino IDE settings:

    Board: Feather HUZZAH ESP8266
    Flash Size: 1M (64k SPIFFS)
    lwIPVariant: v2 Lower Memory
    VTables: Flash
    CPU Frequency: 80MHz
    Upload Speed: 230400
    Erase Flash: Only Sketch
    Port: dev/ttyUSB0, dev/ttyUSB1

    Board: LOLIN (WEMOS) D1 mini Lite
    Flash Size: 1M (64k SPIFFS)
    lwIPVariant: v2 Lower Memory
    VTables: Flash
    CPU Frequency: 80MHz
    Upload Speed: 230400
    Erase Flash: Only Sketch
    Port: dev/ttyUSB0, dev/ttyUSB1

  2018.12.02 - updated new soldered rain sensor board hooked up to a Davis 6465 Rain guage
  https://www.davisinstruments.com/product_documents/weather/spec_sheets/6463_6465_SS.pdf
  
  ---------------------------------------------------------------
*/

#include <FS.h>           //this needs to be first, or it all crashes and burns...
#include <ESP8266WiFi.h>  //https://github.com/esp8266/Arduino
#include <DNSServer.h>
#include <WiFiManager.h>  //https://github.com/tzapu/WiFiManager
#include <ESP8266WebServer.h> //Local WebServer used to serve the configuration portal
#include <ArduinoJson.h>  //https://github.com/bblanchon/ArduinoJson
#include <Adafruit_Sensor.h>  // Adafruit Unified Sensor Driver - https://github.com/adafruit/Adafruit_Sensor

[... 419 lines cut ...]

void setRainTopics() {

  strcat(rssi_topic, topic_loc); strcat(rssi_topic, topic_device); strcat(rssi_topic, chipID); strcat(rssi_topic, "/"); strcat(rssi_topic, "device/"); strcat(rssi_topic, rssi_base);
  strcat(publ_topic, topic_loc); strcat(publ_topic, topic_device); strcat(publ_topic, chipID); strcat(publ_topic, "/"); strcat(publ_topic, "device/"); strcat(publ_topic, publ_base);
  strcat(subs_topic, topic_loc); strcat(subs_topic, topic_device); strcat(subs_topic, chipID); strcat(subs_topic, "/"); strcat(subs_topic, "device/"); strcat(subs_topic, subs_base);

  strcat(rain_topic, topic_loc); strcat(rain_topic, topic_device); strcat(rain_topic, chipID); strcat(rain_topic, "/"); strcat(rain_topic, "rainguage/"); strcat(rain_topic, rain_base);

  Serial.println("concatenated topics: ");
  Serial.println(rssi_topic); // Serial.println(" = " + strlen(rssi_topic));
  Serial.println(publ_topic); // Serial.println(" = " + strlen(publ_topic));
  Serial.println(subs_topic); // Serial.println(" = " + strlen(subs_topic));
  Serial.println(rain_topic); // Serial.println(" = " + strlen(rain_topic));

  Serial.println("");
}

// ========================== set up the topics for this device
void setChipId() {
  snprintf(chipID, 10, "%ld", ESP.getChipId());
  Serial.println(""); Serial.print("Device chip id: "); Serial.println(chipID);
  // return chipID;
}
```

tablatronix, 2019-01-08:
I use the development branch DEV example as my template for sample code, I have also added some better examples of use cases.

code is a bit to read, and not syntax highlighted

tablatronix, 2019-01-08:
It really all depends on flow, I would personally only start cp on demand or if creds empty

Jeppedy, 2020-09-17:
I love that you worked to clean this up.  I may do something similar in my code.
I had a comment/suggestion/question...
If "ParamSave callback" is not called, should you pull back the values from the portal?
If the callback wasn't invoked, doesn't that mean the user did not intend the values to be stored?
I'm still working on some mods to this library, but that's one of the changes I'm proving out within my application.

Your sketch code I'm referring to is:

//JAH: Should these two lines be moved within the IF?
 strcpy(mqtt_server, custom_mqtt_server.getValue());
  strcpy(topic_loc, custom_topic_loc.getValue());

  // ----------------------------- save the custom parameters to FS
  if (shouldSaveConfig) {
    saveConfigFile();
    shouldSaveConfig = false;
  }

tablatronix, 2020-09-18:
Avoid doing work inside a IF or callback, only set a flag etc.

That is why the shouldSaveConfig flag is used

Jeppedy, 2020-09-18:
I'm unclear.  I'm not familiar with this approach of doing work you don't need to do to keep logic out of an IF block.  
If the config is not to be "saved", do you want to pull back the values from the portal?  (That may be the real question here)
If no, then this is logic that is only needed when a certain condition is true, use the IF block.
It takes up cycles, and potentially causes undesirable results later...  Wouldn't it?

ramanraja, 2020-09-19:
I have made similar modification in my code as @Jeppedy has done, and so
far it seems to work.
if  the flag shouldSaveConfig is not true, I just return from the function.
Please let me know if there is any logical or practical flaw in my
approach. That will help correct myself.
(I may not fully understand memory management and the overheads of strcpy
etc.)
Thanks,
Rajaraman
India

tablatronix, 2020-09-19:
sorry I misunderstood, I thought you mean "interrupt funciton"  not if statement