tzapu/WiFiManager #804: Problem with special character in password

giulianetti, 2019-01-10:
### Basic Infos

#### Hardware
**WiFimanager Branch/Release:**
- [ ] Master
- [x ] Development

**Esp8266/Esp32:**
- [ ] ESP8266
- [x] ESP32

**Hardware:			ESP-12e, esp01, esp25**
- [ ] ESP01
- [ ] ESP12 E/F/S (nodemcu, wemos, feather)
- [ ] Other

**ESP Core Version:	2.4.0, staging**
- [ ] 2.3.0
- [ ] 2.4.0
- [ ] staging (master/dev)

### Description

I am have problem when I use special characters in the passwords. 
Exemple: "32<@werpW&uqyW=dmk0FSD"
Could anybody help me?

### Debug Messages

*WM: [2] ESP32 event handler enabled 
*WM: [2] Connecting as wifi client... 
*WM: [2] setSTAConfig static ip not set 
*WM: [3] WIFI station disconnect 
*WM: [1] Connecting to saved AP: Leonardo
*WM: [3] WiFi station enable 
*WM: [2] 30000 ms connectTimeout set
*WM: [2] 30000 ms timeout, waiting for connect...
*WM: [2] Connection result: WL_DISCONNECTED
*WM: [3] lastconxresulttmp: WL_IDLE_STATUS
*WM: [3] lastconxresult: WL_DISCONNECTED
*WM: [1] AutoConnect: FAILED 
*WM: [2] AccessPoint set password is VALID 
*WM: [1] password 
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  AutoConnectAP
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] Config Portal Running, blocking, waiting for clients... 
*WM: [2] NUM CLIENTS: 0

tablatronix, 2019-01-10:
in what password?

tablatronix, 2019-01-10:
ok, let me check it out, must be an encoding issue in the form submission

tablatronix, 2019-01-10:
I dont know that this is a wm problem
`*WM: [1] using password: 32<@werpW&uqyW=dmk0FSD`

are you sure esp supports this password, can you connect manually?

giulianetti, 2019-01-10:
Yes, I can.

See the example bellow:

Code:

`/*
 *  This sketch sends data via HTTP GET requests to data.sparkfun.com service.
 *
 *  You need to get streamId and privateKey at data.sparkfun.com and paste them
 *  below. Or just customize this script to talk to other HTTP servers.
 *
 */

#include <WiFi.h>

const char* ssid     = "Leonardo";
const char* password = "32<@werpW&uqyW=dmk0FSD";

const char* host = "data.sparkfun.com";
const char* streamId   = "....................";
const char* privateKey = "....................";

void setup()
{
    Serial.begin(115200);
    delay(10);

    // We start by connecting to a WiFi `

Debug 

'Connecting to Leonardo
.......................................
WiFi connected
IP address: 
192.168.43.154
connecting to data.sparkfun.com
Requesting URL: /input/....................?private_key=....................&value=1
HTTP/1.1 302 Found
Cache-Control: no-cache
Content-length: 0
Location: https://data.sparkfun.com/input/....................?private_key=....................&value=1
Connection: close'

giulianetti, 2019-01-10:
I am trying to debug the password that the program is reading, however I am having difficulty finding the password variable.
Can you tell me what it is?

bool WiFiManager::wifiConnectDefault(){
  bool ret = false;
  DEBUG_WM(F("Connecting to saved AP:"),WiFi_SSID());
  DEBUG_WM(F("Password:"), **??????**);
  ret = WiFi_enableSTA(true,storeSTAmode);
  if(!ret) DEBUG_WM(DEBUG_ERROR,"[ERROR] wifi enableSta failed");
  ret = WiFi.begin();
  if(!ret) DEBUG_WM(DEBUG_ERROR,"[ERROR] wifi begin failed");
  return ret;
}

tablatronix, 2019-01-10:
so you did begin(ssid,password);

tablatronix, 2019-01-10:
3717e323b9f524c4018cc883350db8072b5d3ff3

giulianetti, 2019-01-11:
Hi tablatronix,

As you can see bellow, the problem is in the character "=", per some reason it is scaping.
I belive that the problem happen when we are saving the new ap (form web).

Debbug

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:952
load:0x40078000,len:6084
load:0x40080000,len:7936
entry 0x40080310
*WM: [1] AutoConnect 
*WM: [2] ESP32 event handler enabled 
*WM: [2] Connecting as wifi client... 
*WM: [2] setSTAConfig static ip not set 
*WM: [3] WIFI station disconnect 
*WM: [1] Connecting to saved AP: Leonardo
`*WM: [3] Using Password: 32<@werpW&uqyW`
*WM: [3] WiFi station enable 
*WM: [2] 60000 ms connectTimeout set
*WM: [2] 60000 ms timeout, waiting for connect...
*WM: [2] Connection result: WL_DISCONNECTED
*WM: [3] lastconxresulttmp: WL_IDLE_STATUS
*WM: [3] lastconxresult: WL_DISCONNECTED
*WM: [1] AutoConnect: FAILED 
*WM: [2] AccessPoint set password is VALID 
*WM: [1] password 
*WM: [3] WIFI station disconnect 
*WM: [3] WiFi station enable 
*WM: [2] Disabling STA 
*WM: [2] Enabling AP 
*WM: [1] StartAP with SSID:  AutoConnectAP
*WM: [1] AP IP address: 192.168.4.1
*WM: [3] setupConfigPortal 
*WM: [1] Starting Web Portal 
*WM: [3] dns server started with ip:  192.168.4.1
*WM: [2] HTTP server started 
*WM: [2] WiFi Scan ASYNC started 
*WM: [2] Config Portal Running, blocking, waiting for clients... 
*WM: [2] NUM CLIENTS: 0

mindforger, 2019-01-11:
probably because it is a 
>form method='get'

i did not look into the parser deep enough but this may be misread as an additional key/value pair from the webserver

giulianetti, 2019-01-11:
I solved this problem changing the file strings_en.h, as bellow:

Before:
`const char HTTP_FORM_START[]       PROGMEM = "<form method='POST' action='{v}'>";`

After:
`const char HTTP_FORM_START[]       PROGMEM = "<form method='POST' enctype=\"multipart/form-data\" action='{v}'>";`

mindforger, 2019-01-11:
Huh? What file was i reading then?

tablatronix, 2019-01-11:
hmm interesting, I guess some browsers automatically handle the enctype
or maybe you dont have the newest branch updates with the <meta charset='UTF-8'> ?

acidobinario, 2019-03-05:
seems like a base64 conversion of the password (maybe with js? ) would help (but you'll have to decode it just before this lines:
```
void WiFiManager::handleWifiSave() {
  DEBUG_WM(F("WiFi save"));

  //SAVE/connect here
  _ssid = server->arg("s").c_str();
  _pass = server->arg("p").c_str();
...
```
)

amites, 2019-03-22:
tried added encoding -- still failing for me

about to tie in the base64 encode/decode -- my current guess is that the strings are being url encoded and not url decoded before being implemented though I've yet to really crawl through to verify that...

acidobinario, 2019-03-23:
@amites where's the code that you've used? i'll like to see it.

tablatronix, 2019-03-23:
your still having issues with this?

I added debugging, what do you logs say ?

amites, 2019-03-23:
I forked the repo at https://github.com/amites/WiFiManager/tree/cmidgley
the last 4 commits are me tweaking the form output

All I committed so far was added enc-type coding to the different forms, and monitored the serial output
it seems to freeze when trying to connect until about 10 minutes of trying at which point it seems to memory dump and reset itself trying to autoconnect with the previously saved creds

I'm pretty green to C++ so wasn't sure how to add the base64 decode referred to in the above issue 
link -- have code ready to push that adds base64 encoding to the form before submitting that I'd be happy to share 

I can see that special characters are URL encoded as part of the `GET` request and I suspect they aren't being decoded before being used as credentials to login to wifi

tablatronix, 2019-03-23:
I posted above that I could not reproduce so it must be a browser dependant issue. I also added debugging to clearly show what was received. 

Not sure what else can be done, charaet encoding should work on all browsers

tablatronix, 2019-03-23:
Ok are you not using development , this issue is for development and fixes are applied there only atm

tablatronix, 2020-05-11:
I will close this if it is not reproducible. let me know if it needs to be tested more.

closed 2020-12-16 by tablatronix