tzapu/WiFiManager #806: config portal  - Opens msn.com - With Computer Ethernet Connected

froggerXR7, 2019-01-14:
config portal  - Opens msn.com - With Computer Ethernet and WiFi Connected

Wifi Manager Works Good For Me For A Long Time BUT.

I have been using wifi for ever 
Just The other day i plug in my laptop ether cable and discovered by chance that WifiManager When connecting to the AP It Did Not Open The Manager Config Portal ?  It Opened msn com ?

Laptop Has Both WIFI AND Ether Cable Connected

Works Good When Ether Cable Is Disconnected 
This issue Dose Not Bother Me I rarely use ether net

mindforger, 2019-01-14:
you seem to suffer from an issue with blocking specific websites of microsoft, this seems to be a known behaviour and has probably nothing to do with the device

https://www.bleepingcomputer.com/forums/t/612201/msncom-opens-upon-plugging-in-ethernet-software-bug-virus-or-network-issue/

https://www.reddit.com/r/microsoft/comments/39yhp0/msn_homepage_keeps_opening_up_in_my_browser_chrome/

tablatronix, 2019-01-14:
when using a captive portal you should disconnect other networks if you have this issue

mindforger, 2019-01-14:
Workarounds are probivided in the linked articles, if you meant my post :)

tablatronix, 2019-01-14:
-[ ] add note to readme, sometimes data must be disabled to use captive portal, this is pretty standard and most wifi IOT devices will ask this.